d-ronin/dRonin #1420: flyingpio: ppm is broken

mlyle, 2016-10-09:
Failsafes; intermittently flashes correct values according to @mhv-shared from IRC.  Have not tried myself.

xfce, 2017-02-27:
i change the pios_flyingpio.c's cmd->receiver_protocol = HWSHARED_PORTTYPES_SBUS
to cmd->receiver_protocol = HWSHARED_PORTTYPES_PPM
then no change, still only found the throttle stick in GCS's wizard ,
try both quitoxe and atifice branch

jihlein, 2018-02-03:
I've been looking into this and here's what I've found so far:
1)Made the changes in #2043 to correct RTC frequency.  Verified the RTC frequency with a Logic Analyzer to be 625 Hz.  PPM still does not work correctly.
2)Made a new time base definition TIM14 with the TIM period = 0xFFFF.  PPM still does not work correctly.
3)Changed the processor clock from 40 MHz to 48 MHz.  PPM still does not work correctly.

Interesting things to note:
1)After the above changes, PPM will sometimes lock, after waiting a while,  This wait has been observed to be as short as 56 seconds, or as long as 203 seconds, so it's not consistent.  And sometimes I've waited 5 plus minutes with no lock.  I'm not sure if all of the changes, some of the changes, or one of the changes above are responsible for this.
2)Once locked, however, I can cycle the RX power off and on with immediate lock recovery, and I can do this multiple times with no failure to immediately lock.
3)The PWM frame time in the code is 20 Msec.  I'm using a D4R-II with a 27 mSec frame time.  Don't know if this matters too much as it works fine on the other target boards I've tried, but it might be worth experimenting with.

To be continued..... :)

mlyle, 2018-03-24:
Confirmed working in #2140

closed 2018-03-24 by mlyle