d-ronin/dRonin #1428: DSMX Satellite Sparky 2 - No RC Input

jonnymears, 2016-10-15:
* Seen in version:  "Quixote" 
* On flight controller:  Sparky 2 Clone 

Issue details:

I have a clone Sparky 2 flashed with quixote that I can't get to work with a spektrum satellite.  The satellite is bound and connects to my transmitter.  When looking at the input page, I get the right values on all my channels (1-6).  All Channels respond appropriately when I use the transmitter.  But the wizard wont detect any inputs, and I get a "No RC input" error.  I am able to manually calibrate, but as soon as I set an arm mode the no rc input failsafe kicks on and prevents me from flying.  The receiver port is putting out the right voltage at 3.3v.  I have tried the receiver port, flexiport and main port.  All get the same problem.  

I have 2 other quads with cc3d boards on the same version of Dronin that are not having the issue with the same satellite receiver.  

Anyone seen this before?  

Thanks!

jonnymears, 2016-10-15:
I'm using a Devo 10 with deviation.  End points set to +-  100%.  Bound in dsmx.  On the input page I am getting a max of 1726 mid 1024 and min 326.  Thanks for the fast response!

tracernz, 2016-10-15:
That sounds good. So it sounds like just the arm channel is playing up. Are you sure it's got the same limits?

mlyle, 2016-10-28:
You could also try setting DSM to force 11 bits.  Closing for inactivity.

closed 2016-10-28 by mlyle