d-ronin/dRonin #1433: FlyingPi: may need to drive STM reset line harder

mhv-shared, 2016-10-17:
* Seen in version (release version or githash): next
* On flight controller (if applicable): flyingpi

Issue details:
I've been seeing intermittent CRC errors leading to lockup of the simulator. Realized they look a lot like the STM is resetting, and I noticed that after running the OpenOCD `flyingpi.cfg` script, the pin driving reset (BCM17) has been reconfigured as an input. Since the pullups on the pi are pretty weak, I think noise might be triggering resets.

mlyle, 2016-10-17:
Ahhh.  Thank you.

mlyle, 2016-10-17:
In the meantime I suggest this shell script:

```
echo 17 > /sys/class/gpio/export
echo "out" >/sys/class/gpio/gpio17/direction
echo 1 >/sys/class/gpio/gpio17/value
```