d-ronin/dRonin #1438: simulator dies when receiving NMEA GPS data

mhv-shared, 2016-10-19:
* Seen in version (release version or githash): next
* On flight controller (if applicable): simulator/flyingpi

Issue details: when configured with `GPSDataProtocol=NMEA`, the simulator silently exits almost immediately. This only occurs if the GPS is actually connected. Works fine when set to UBX.

mlyle, 2016-10-20:
Heh this is somewhat likely to affect more than flyingpi, given that almost no one uses NMEA.

tracernz, 2016-10-20:
I thought that too. I have an NMEA GPS on a quad, so I'll give it a try on a regular target in the next couple of days.

mlyle, 2016-10-20:
that might work too yes :D

mlyle, 2018-03-24:
No repro on bench testing in flightd/sim, though I found a lot of minor things to fix.

tracernz, 2018-03-24:
Repro with MTK3329:
```
(gdb) run
Starting program: /home/mike/dev/dronin/build/flightd/flightd -S gps:/dev/ttyUSB0
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/usr/lib/libthread_db.so.1".
Beginning simulation environment
[New Thread 0x7ffff74b6700 (LWP 15799)]
Initialization task running
Started thread (init) p=0x5555557df260
HW serial number-- hex: 000000000000000000000000
[New Thread 0x7ffff6cb5700 (LWP 15800)]
Started thread (pios_rtc_task) p=0x7ffff0000b20
Pseudo-RTC started
[New Thread 0x7ffff64b4700 (LWP 15801)]
Started thread (pios_tcp_rx) p=0x7ffff00056a0
tcp dev 0x7ffff0004e30 - socket 5 opened - result 0
Completed PIOS_Board_Init
[New Thread 0x7ffff5cb3700 (LWP 15802)]
Started thread (pios_serial_rx) p=0x7ffff00065a0
serial dev 0x7ffff0005d60 - fd 6/6 opened
Setting Serial ID 0x7ffff0005d60 to 9600 bps
serial dev 0x7ffff0005d60 - (path /dev/ttyUSB0)
SimSensorsInitialize: Using simulated sensors.
SimSensorsInitialize: starting SIMULATED sensors

Thread 2 "init" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7ffff74b6700 (LWP 15799)]
0x00005555555a7c8a in PIOS_COM_validate (com_dev=0xf0006700)
    at /home/mike/dev/dronin/flight/PiOS/Common/pios_com.c:69
69		return (com_dev && (com_dev->magic == PIOS_COM_DEV_MAGIC));
(gdb) bt
#0  0x00005555555a7c8a in PIOS_COM_validate (com_dev=0xf0006700)
    at /home/mike/dev/dronin/flight/PiOS/Common/pios_com.c:69
#1  PIOS_COM_ChangeBaud (com_id=4026558208, baud=19200)
    at /home/mike/dev/dronin/flight/PiOS/Common/pios_com.c:228
#2  0x0000555555571c0d in updateSettings ()
    at /home/mike/dev/dronin/flight/Modules/GPS/GPS.c:331
#3  GPSInitialize () at /home/mike/dev/dronin/flight/Modules/GPS/GPS.c:146
#4  0x0000555555591cba in initTask (parameters=<optimized out>) at main.c:118
#5  0x00007ffff787508c in start_thread () from /usr/lib/libpthread.so.0
#6  0x00007ffff75ace7f in clone () from /usr/lib/libc.so.6
```

The com_dev pointer is invalid.

mlyle, 2018-03-24:
Right.. that's the issue I just fixed in the PR on amd64, but it wouldn't affect rpi which is 32bit.

tracernz, 2018-03-24:
Ahhh, right. Same problem on aarch64, but now re-reading OP, this is NMEA specific which is weird.

mlyle, 2018-03-24:
Yah, it's pretty weird.  Probably should close this, though, since it works (and we don't build a sim bin currently for aarch64 / would really not encourage someone to fly that until we've audited a lot more 64 bit path stuff).

mlyle, 2018-04-18:
No repro, no objections, closing.

closed 2018-04-18 by mlyle