d-ronin/dRonin #1442: micro scisky, no ppm

lolorc, 2016-10-25:
* Seen in version (release version or githash): ALL
* On flight controller (if applicable): micro scisky v1.2

Issue details:
they've updated the micro scisky, the receiver now does DSMX and outputs PPM, there's a MPU6050, the board is seen as a REV6 naze32.

I've never managed to make it work with dR, ppm input stays at 65535.
I've tried with 20160720.1, 20160409.2, 20160409.1 and next

Before declaring the board DOA, i tried CF. and it works with RX_PPM.
it works with latest CF and also with a very old version (v1.2)

I've tried to alter the ppm input pin L814 in board_hw_defs.c :
.channels = &pios_tim_rcvrport_all_channels[X], with X  from 0 to 7

looking at previous issues with ppm, i tried disabling SERVO just in case, but i guess it's useless since I only use rcvrport=PPM, rcvrserial=DISABLED

is there something I am missing ? what could I do to provide more info ?

lolorc, 2016-10-25:
[scisky.txt](https://github.com/d-ronin/dRonin/files/550321/scisky.txt)

mlyle, 2016-10-25:
I don't know, like I said on IRC.  Without having the board on hand...

I don't think we'll be able to support every single clone variant of Naze.

closed 2016-10-25 by mlyle

lolorc, 2016-10-26:
I just opened the issue as a way to gather information so i can try to fix the issue myself.
i specifically bought this old F1 brushed board because it was marked as supported by dronin, otherwise I would have gone a for one of the  F3 with a frsky rx
I'll try to persevere !