d-ronin/dRonin #1454: Flight[ChibiOS]: Clock init is ineffective since PLL is running after bootloader

tracernz, 2016-11-07:
ChibiOS function `stm32_clock_init` attempts to setup the PLL, assuming that it is inactive. PLL is usually already activated by the bootloader when we jump to firmware. Writes to PLL configuration registers are ineffective when the PLL is active. Relates to #1448.

mlyle, 2016-11-07:
Yes.  Arguably this is WONTFIX/SHOULDNTFIX.

Question is, is there an anticipated benefit from de-initing the PLL?  I'm not even 100% comfortable with toggling back to internal clock and then back to PLL.

tracernz, 2016-11-07:
The best thing I think would be to properly integrate ChibiOS so it shares ALL of the same clock/PLL settings as bootloader. That will minimise the chance of anything bad happening due to this.

PR #2123 (2018-03-18): flight: cleanup clock handling and some makefile stuff
1. ensure that it is possible to override pios includes in the board-info directory (we don't currently need this, but for awhile on this PR I did, and I think it's a good/related thing to have).
2. ensure that in cmsis_system we comply with datasheet recommendations on PLL configuration (namely, on 8MHz F4 targets, don't do / 10 * 420 / 2, but instead do / 8 * 336 / 2, which keeps the PLL divided reference above 1MHz).
3. factor chconf common, and make sure the values are the same as cmsis_system.  also disable clock init.  Fixes #1454
closed 2018-03-18 by tracernz