d-ronin/dRonin #1458: ESC Motor imput on Naze32 Rev6 & LUX

Duanevc, 2016-11-16:
* Seen in version (release version or githash): 
* On flight controller (if applicable): Naze32 rev6 & LUX

Issue details: Hi Guys
When i go through my setup, im selecting the ONESHOT42 for my esc's. When all is said and done i try to do test hover but my one motor keep on spinning faster and faster. I tried the Calibrate ESC function in dRonin with every possible option but it still remain to do the same.

I Even Calibrated my ESC's with BL_Heli suite to make sure all my ESC's are in Sync. 

Same setup Using CleanFlight and BetaFlight with No issues. I hate CF and BF and have been flying with dRonin for ever. And would like to stay with dRonin.

Same issue with a New LUX board i bought a few days ago.

Any comments will be appreciated.

dustin, 2016-11-16:
Does this happen with regular oneshot? (125-250)  I don't see why it would, but oneshot42 isn't really advised.

Duanevc, 2016-11-16:
Im sure ive tried  Oneshot 125... I'll just double check

jihlein, 2016-11-16:
The maximum loop rate in Dronin on Naze32 is 500 Hz, as determined by the MPU rate setting.  For the Lux, it is 1000 Hz.  Oneshot 125 maximum pulse width of 250 uSec is capable of 4 kHz update, 4 times faster than the max loop rate you are capable of achieving.   There's no point to Oneshot 42, it's lower resolution and more susceptible to noise.

Do the motors behave normally when test motor outputs is selected on the output pane of the GCS?

Duanevc, 2016-11-16:
Yes the motors behave normally when armed, just when i give throttle input in goes all wrong. Cant even hover to Auto tune. The motors have a bad twitch in them when inputs are given

closed 2017-01-28 by mlyle