d-ronin/dRonin #1462: Config gadget not loading when building as Debug on Windows

yamahabest, 2016-11-19:
* Seen in version (release version or githash): Next
* On flight controller (if applicable): n.a.

Issue details: Low-low-low priority.
When building as Debug (default) on Windows, on startup the GCS can't load Configd.dll: "A dynamic link library (DLL) initialization routine failed". It builds fine though.

When building as Release, it works without loading errors. (After adding the Config gadget to the Configuration screen again).

glowtape, 2017-02-19:
Same happens here. I used Process Monitor as per recommendation of @tracernz (Filemon and Regmon are deprecated) to track this down to something.

There aren't any DLL loading failures from the operating system side, all Load Image operations finish in successes, so it's not like a dependency doesn't resolve. Nothing special in regards to registry keys. However there's some odd stuff happening with Configd.dll, that doesn't happen with random other plugin DLLs, like Dronind.dll (which I presume is an important component). Not sure what to make of it. 

See the attachments, specifically the calls to QueryNameInformationFile that happen for Configd.dll but not Dronind.dll. In the release configuration, they don't happen for Config.dll either.

[image]

[image]

Not really a ball breaking bug, as long you know how to compile the release version.

tracernz, 2017-03-10:
Captured a stack trace of the failure I think (with Microsoft's "Debug Diagnostic Tool":
```
  P:\dronin\build\ground\gcs\lib\dr\plugins\dRonin\UAVObjectsd.dll loaded at 0x5d2f0000
  P:\dronin\build\ground\gcs\lib\dr\plugins\dRonin\AeroQuadd.dll loaded at 0x5d280000
  P:\dronin\build\ground\gcs\lib\dr\plugins\dRonin\UAVObjectUtild.dll loaded at 0x5d250000
  P:\dronin\build\ground\gcs\lib\dr\plugins\dRonin\UAVSettingsImportExportd.dll loaded at 0x5d210000
  P:\dronin\tools\Qt5.8.0\5.8\msvc2015\bin\Qt5XmlPatternsd.dll loaded at 0x5cb60000
  P:\dronin\build\ground\gcs\lib\dr\plugins\dRonin\UAVTalkd.dll loaded at 0x5cb10000
  P:\dronin\build\ground\gcs\lib\dr\plugins\dRonin\UAVObjectWidgetUtilsd.dll loaded at 0x5ca90000
  P:\dronin\build\ground\gcs\lib\dr\plugins\dRonin\BrainFPVd.dll loaded at 0x5ca10000
  P:\dronin\build\ground\gcs\lib\dr\plugins\dRonin\Brotronicsd.dll loaded at 0x5c900000
  P:\dronin\build\ground\gcs\lib\dr\plugins\dRonin\Configd.dll loaded at 0x5c320000
  P:\dronin\build\ground\gcs\bin\Qwtd.dll loaded at 0x5c160000
  P:\dronin\tools\Qt5.8.0\5.8\msvc2015\bin\Qt5PrintSupportd.dll loaded at 0x5c0d0000
  C:\WINDOWS\SysWOW64\COMDLG32.dll loaded at 0x77910000
  C:\WINDOWS\SysWOW64\WINSPOOL.DRV loaded at 0x70ad0000
  C:\WINDOWS\WinSxS\x86_microsoft.windows.common-controls_6595b64144ccf1df_6.0.14393.447_none_89c64d28dafea4b9\COMCTL32.dll loaded at 0x71b60000
  C:\WINDOWS\SysWOW64\bcrypt.dll loaded at 0x72ba0000
  Exception 0XC0000005 on thread 1204. DetailID = 1
  Stack Trace
ChildEBP RetAddr  Args to Child              
017fe5ac 5fa7abf5 5c8a0818 5c5e9394 00000000 Qt5Cored!objectClassName+0x12 [c:\users\qt\work\qt\qtbase\src\corelib\kernel\qmetaobject.cpp @ 306]
017fe5cc 5c3c8cb6 017fe610 5c5e9394 00000000 Qt5Cored!QMetaObject::tr+0x25 [c:\users\qt\work\qt\qtbase\src\corelib\kernel\qmetaobject.cpp @ 375]
017fe5e8 5c33090a 017fe610 5c5e9394 00000000 Configd!ConfigInputWidget::tr+0x26 [p:\dronin\ground\gcs\src\plugins\config\configinputwidget.h @ 58]
017fe6e0 5f449669 5c3308c0 5c5d310c f05bff1f Configd!`dynamic initializer for 'ConfigInputWidget::armingMethods''+0x4a [p:\dronin\ground\gcs\src\plugins\config\configinputwidget.cpp @ 74]
WARNING: Stack unwind information not available. Following frames may be wrong.
017fe6f4 5c599cc1 5c5d3000 5c5d32e0 f37edddc ucrtbased!initterm+0x49
017fe738 5c599b69 5c320000 00000000 00000001 Configd!dllmain_crt_process_attach+0xe1 [f:\dd\vctools\crt\vcstartup\src\startup\dll_dllmain.cpp @ 65]
017fe750 5c599f53 5c320000 00000001 00000000 Configd!dllmain_crt_dispatch+0x29 [f:\dd\vctools\crt\vcstartup\src\startup\dll_dllmain.cpp @ 133]
017fe794 5c59a13f 5c320000 00000001 00000000 Configd!dllmain_dispatch+0x93 [f:\dd\vctools\crt\vcstartup\src\startup\dll_dllmain.cpp @ 190]
017fe7a8 77d6e58e 5c320000 00000001 00000000 Configd!_DllMainCRTStartup+0x1f [f:\dd\vctools\crt\vcstartup\src\startup\dll_dllmain.cpp @ 248]
017fe7c8 77d40e46 5c328a8a 5c320000 00000001 ntdll!RtlDecompressBuffer+0xee
017fe818 77d2aafc 00000001 00000000 d8f7298a ntdll!LdrShutdownThread+0x386
017fe898 77d2a99b 07ba3958 00000000 017fe954 ntdll!LdrShutdownProcess+0x3ac
017fe8bc 77d30fb6 017fe8d7 017fe958 017fe954 ntdll!LdrShutdownProcess+0x24b
017fe8d8 77d4029b d8f72832 017fea98 017feb28 ntdll!RtlCreateUnicodeString+0x126
017fe920 77d3cd94 00000600 00000004 00000000 ntdll!RtlEnterCriticalSection+0x37b
017fea6c 77d2a64f 00000000 00000001 017fea90 ntdll!RtlpCreateProcessRegistryInfo+0x3d4
017feaf0 77ac9538 00000001 017feb18 017feb28 ntdll!LdrLoadDll+0x7f
017feb30 77acc711 01948af0 00000000 00000000 KERNELBASE!LoadLibraryExW+0x138
017feb44 5fa688d6 01948af0 cf169926 017feb6c KERNELBASE!LoadLibraryW+0x11
017fee44 5fa65d83 cf1699de 017fee6c 5f7d0d6c Qt5Cored!QLibraryPrivate::load_sys+0x1c6 [c:\users\qt\work\qt\qtbase\src\corelib\plugin\qlibrary_win.cpp @ 102]
017feebc 5fa65f3b cf16999e 00000008 01948468 Qt5Cored!QLibraryPrivate::load+0x63 [c:\users\qt\work\qt\qtbase\src\corelib\plugin\qlibrary.cpp @ 538]
017feefc 5fa59a0a 017fefa4 017fef20 017fefa4 Qt5Cored!QLibraryPrivate::loadPlugin+0x5b [c:\users\qt\work\qt\qtbase\src\corelib\plugin\qlibrary.cpp @ 591]
017fef0c 62014bac cd153c66 017fef00 017fef10 Qt5Cored!QPluginLoader::load+0x8a [c:\users\qt\work\qt\qtbase\src\corelib\plugin\qpluginloader.cpp @ 240]
017fefa4 6200da63 cd152382 017ff04c 00f14000 ExtensionSystemd!ExtensionSystem::Internal::PluginSpecPrivate::loadLibrary+0x16c [p:\dronin\ground\gcs\src\libs\extensionsystem\pluginspec.cpp @ 786]
017ff040 6200d350 0199de18 00000003 cd1522c6 ExtensionSystemd!ExtensionSystem::Internal::PluginManagerPrivate::loadPlugin+0x2d3 [p:\dronin\ground\gcs\src\libs\extensionsystem\pluginmanager.cpp @ 655]
017ff104 6200c2c2 017ff678 017ff7a4 001f9ef3 ExtensionSystemd!ExtensionSystem::Internal::PluginManagerPrivate::loadPlugins+0x150 [p:\dronin\ground\gcs\src\libs\extensionsystem\pluginmanager.cpp @ 551]
017ff110 001f9ef3 af3481c0 00000007 002172fc ExtensionSystemd!ExtensionSystem::PluginManager::loadPlugins+0x12 [p:\dronin\ground\gcs\src\libs\extensionsystem\pluginmanager.cpp @ 249]
017ff7a4 0020d68e 00000001 018c4378 018c9fe8 drgcs!main+0x2ee3 [p:\dronin\ground\gcs\src\app\main.cpp @ 384]
017ff7b8 0020d510 af348e74 001f11f4 001f11f4 drgcs!invoke_main+0x1e [f:\dd\vctools\crt\vcstartup\src\startup\exe_common.inl @ 64]
017ff810 0020d3ad 017ff820 0020d6a8 017ff834 drgcs!__scrt_common_main_seh+0x150 [f:\dd\vctools\crt\vcstartup\src\startup\exe_common.inl @ 253]
[... cut ...]
  Thread exited. Exiting thread - System ID: 9436. Exit code - 0xffffffff
  Thread exited. Exiting thread - System ID: 1644. Exit code - 0xffffffff
  Process exited. Exit code - 0xffffffff

***********************
*  EXCEPTION DETAILS  *
***********************

DetailID = 1
	Count:    2
	Exception #:  0XC0000005
	Stack:        

***********************
*  EXCEPTION SUMMARY  *
***********************

	|--------------------|
	| Count | Exception  |
	|--------------------|
	| 2     | 0XC0000005 |
	|--------------------|
```

PR #1621 (2017-03-10): GCS:Config: Fix debug build on Windows
tr causes an exception when initializing armingMethods... not worth the effort to fix since nobody has been forthcoming with translations anyhow

Fixes #1462
closed 2017-03-10 by mlyle