d-ronin/dRonin #1464: Dronin dshot 150 300 600

sleepyfpv, 2016-11-21:
* On flight controller (if applicable): brainre1 ,BrainFPV, others(?).

Issue details:

Is there anyone here that could implement dshot 150 300 600 to dronin ?

tracernz, 2016-11-21:
No plans at this stage due to the lack of evidence of any actual improvement vs. Effort required and restrictions on other features due to hardware allocations.

mluessi, 2016-11-22:
I wonder if flyduino did something similar to when they introduced OneShot, i.e., D-Shot enables some other improvement in the fw that translates to better flight performance (like when they enabled braking when OneShot was enabled).

jhitesma, 2016-11-23:
Well, oneshot didn't eat resources as bad as dshot.  And the benefits of oneshot were measurable - heck Peabody's tests with autotune were the first measurements proving a benefit from oneshot.  And I then tested oneshot with and without braking to show how much bigger the benefit from braking was compared to the relatively minor (but still measurable) difference from oneshot when blheli added support for oneshot and it was then possible to test braking and oneshot separately.  

I wonder if a bigger reason people are experiencing a perceived benefit from dshot is they're getting calibration right (based on how many people I've seen mess that up royally) and how many are coming from multishot where they're possibly having issues with noise and not realizing it.

mlyle, 2016-11-23:
There's also just confirmation bias.  If I change something, and then go looking at a complicated system in a subjective way... odds are I'll feel like I improved something.   This happens over and over-- remember the "reversed motor direction is better" thing?

yamahabest, 2016-11-30:
More in-depth information by @mlyle: https://forum.dronin.org/forum/d/230-dshot-for-the-win/4

PR #1522 (2017-02-01): pios_servo: support dshot
fixes #1464

* add new calls to pios_tim to reinit outputs and to change timer outs
to general purpose output
* add new "inline delay" infrastructure for high resolution
timing/bitbanging requirements
* abuse the actuator rates 65532, 65533, 65534Hz for dshot 300/600/1200
* teach pios_servo how to generate a dshot signal, and factor some
things slightly more cleanly

Todo:

* [x] GCS support
* [x] Verify autotune can work on CC3D with dshot
* [x] Test on dshot ESC
closed 2017-02-01 by tracernz