d-ronin/dRonin #1474: UAVO data missing from autoupdate save

pug398, 2016-12-04:
* Seen in version (release version or githash): 8ccb52cb

* On flight controller (if applicable): sparky

Issue details:
The autoupdater does not save all uavo when upgrading. Manual export does save all uavo.

pug398, 2016-12-06:
Working with this morning's release 3aba6cfb. One of those things :)

mlyle, 2016-12-06:
Yes, it probably is the version you updated from.  If you know the build you were coming from I can check it out.

Basically there's a known issue that if the cloud service doesn't know about the githash you're coming from (like if it is a temporary one from a pull request) it will fall back to a guess as to the version and sometimes can't read all the UAVOs if this guess is bad. 

We tried to make this guess better in #1124---- and it seems like it does, but doesn't cover all the cases.  In any case, it doesn't affect end users; it just tortures us. :p

pug398, 2016-12-06:
Pretty sure it was weekend regular morning build with all the changes in it but will have to check when I get home. All good now :)

mlyle, 2016-12-06:
*nods* important thing is the release you're upgrading from, not the one to.  If you still have that info it'd be helpful.

pug398, 2016-12-06:
Ahh I see now. It could very well have been the test version of new autotune and that would be exactly as you explained it earlier. I just didn't catch on to what you were saying.

mlyle, 2016-12-06:
Yes, I think that fits.

I looked at the new autotune artifacts.

* the githash 9d7879fb that they were built from-- that autotune branch merged onto next -- doesn't exist anymore as github throws that stuff away.  No one had upgraded from 9d7879fb before github threw it away, so the cloud service never learned how to upgrade from it.
* the next-best-ancestor-hash exists.  But since then, ManualControl changed (flight mode names) and autotune stuff.  So that stuff would be lost, even if everything did the right thing.

So I think this is just a scenario we have to live with now and then when on PR branches. :/

closed 2016-12-06 by mlyle