d-ronin/dRonin #1476: GCS not loading default configuration

PTDreamer, 2016-12-05:
* Seen in version 3aba6cfb892348f3825bbd43cb37d19052f0acbc: 

Issue details:
Self built GCS on linux 64 fails to load default configuration after message "Some configuration is missing in the imported config and will be replaced..." and ends up loading with two workspaces that crash if clicked.
To trigger the issue just delete existing config file or start the gcs with --reset-config switch

PTDreamer, 2016-12-05:
@mlyle building again probably my mistake.

mlyle, 2016-12-05:
Very strange.  On Ubuntu 16.10, from a build tree ( ./build/ground/gcs/bin/drgcs ) all 4 confs seem to work for me

mlyle, 2016-12-05:
(Quite likely it's something with incremental builds, if I were to guess)

PTDreamer, 2016-12-05:
Yeh my bad, wrong origin I think.
Sorry!

closed 2016-12-05 by PTDreamer