d-ronin/dRonin #1492: RE1 crashes immediately upon providing S.bus input

mlyle, 2016-12-20:
Reported by user on next; confirmed on seppuku branch.  Don't know if it was the LED changes or what.

mluessi, 2016-12-20:
Did some tests: I can run the setup wizard until the point at which the sticks have to be moved to the maximum extents, then the FC resets. Doesn't seem to be related to #1488, as I can also reproduce it using the version before it merged (d74c5701a).

mluessi, 2016-12-20:
Couldn't reproduce it on sparky2

mlyle, 2016-12-20:
Yah I've been flying next a fair bit (but not on RE1) without issue.  Can try a little to bisect later today.

mluessi, 2016-12-21:
The ManualControl stack was overflowing by 100 bytes. Bump it by 200 bytes? Not sure what caused the stack increase.

glowtape, 2016-12-21:
Must have happened past November 30. I have a next build of that date on my RE1-quad and am using S.Bus without any issues.

PR #1493 (2016-12-21): Bump ManualControl stack by 200 bytes
ManualControl stack was overflowing by about 100 byes on RE1 causing a crash when used with S.Bus. Not sure why the needed stack grew so much.

Fixes #1492
closed 2016-12-21 by mlyle