d-ronin/dRonin #1499: Exporting UAVOs with no board attached potentially confusing

jhitesma, 2016-12-31:
* Seen in version (release version or githash): d0ab1073

Issue details:
Autoupdate failed leaving me in bootloader.  Didn't notice I was in bootloader and tried to export my UAVOs - it "worked" but I got an essentially empty UAVO dump:

<uavobjects>
    <version>
        <hardware revision="9f" serial="000000000000000000000000" type="f0"/>
        <firmware date="" hash="" tag=""/>
        <gcs date="detached at 19" hash="19bd0c53" tag="(HEAD detached at 19bd0c5)"/>
    </version>
    <settings/>
</uavobjects>

Which while not wrong isn't exactly very helpful to the user.  I think this may have happened to me once before and resulted in me trying to restore from an empty dump, and I think it may have happened to another user I spoke with.

Perhaps it would be better if the GCS were to give an error and not allow exporting when there's no actual firmware to dump from?  That way no one can accidentally export an empty dump and then wind up having to start from scratch if they try to restore from it.

mlyle, 2017-03-16:
May be best to grey/ungrey menu item.  Of course, it'll be possible to export as you disconnect or something.. but building a full set of interlocks for this seems excessive.

jhitesma, 2017-03-16:
I was thinking either greying out the option like you suggest, or warning the user after the export if there was no data in it.  Wasn't sure which approach would be easier to implement.

tracernz, 2017-03-17:
Just disable the menu option when telemetry session is not connected?

mlyle, 2017-03-17:
@tracernz yes

tracernz, 2017-06-25:
Broken by #1847

PR #1840 (2017-06-23): GCS:UAVSettingsImportExport: Disable commands when not connected
Fixes #1499
PR #1857 (2017-06-28): GCS:UAVSettingsImportExport: Disable commands when not connected
Fixes #1499 
Resurrects #1840
closed 2017-06-28 by mlyle