d-ronin/dRonin #1518: Brain hanging on boot

dustin, 2017-01-16:
* Seen in version (release version or githash): e55cc1bb09ca16d1fad1c2733d963e2fe396705b
* On flight controller (if applicable): Brain

Issue details:

No OSD at boot.   Can arm, but automatically disarms ~1s into flight.

Note:  I bisected this which pointed to e55cc1bb09ca16d1fad1c2733d963e2fe396705b -- e55cc1bb09ca16d1fad1c2733d963e2fe396705b^ (efb7bc7) is fine.

dustin, 2017-01-16:
I also tried reverting e55cc1bb09ca16d1fad1c2733d963e2fe396705b on b8422e1ace6b5cde4fc63914446d7f095b1e6108 and it seems fine.

mluessi, 2017-01-16:
How can you arm if it hangs on boot? If it hangs, the easiest would probably be to use the debugger to check where it hangs.

dustin, 2017-01-16:
No idea.  It almost works, but no osd and only one second flights.

Apparently I forgot to add that it works fine if I boot via USB before
adding battery power.  I get osd and can fly.  Weird behavior is only when
booting via battery.

mlyle, 2017-01-16:
I tested this exact changeset on brain but I can't say for certain that I didn't power initially by USB.  It will require some further investigation.

mluessi, 2017-01-19:
I did some tests. The brain OSD doesn't work at all on current next and it constantly resets. I tested the version right before the seppuku branch was merged and it's working fine, so definitely some changes in that branch broke it. @mlyle any ideas?

dustin, 2017-01-19:
It's the one change I cited.  I flew Monday with next - that change.

jhitesma, 2017-01-19:
I tried to test this last night...but couldn't remember which port was which on my OG Brain and the support pages are currently unavailable as the cert on them expired in early Nov and the site is configured to use HSTS so Chrome can't fall back to HTTP.  Sent a message on FB to the Brain page but figured I'd mention it here to make sure @mluessi doesn't miss it ;)

dustin, 2017-01-19:
Good to know I didn't waste my time on this.  :)
https://github.com/d-ronin/dRonin/wiki/BrainFPV

PR #1520 (2017-01-22): pios_video/brain: correct pixel timer threshold
May fix #1518, untried
commit 40b14164 (2017-01-22): pios_video/brain: correct pixel timer threshold

May fix #1518, untried
closed 2017-01-22 by tracernz