d-ronin/dRonin #1535: Use QGamePad in GCS

mlyle, 2017-01-30:


tracernz, 2017-01-30:
Good call. :+1:

tracernz, 2017-03-25:
It really is just Q __Gamepad__. Cannot get half the axes and buttons of my joystick out through it even through the undocumented parts of the API so it wasn't capable of controlling GCSControl.

tracernz, 2017-04-07:
Played with this some more, with the Qt 5.9 docs (https://doc-snapshots.qt.io/qt5-5.9/qtgamepad-index.html) and reading the source code (https://github.com/qt/qtgamepad/tree/dev/src). I _can_ get all the axes on my joystick to work with the libevdev backend. :grin: I'll hack things into a workable state and then see how it will work on macOS (GCController) and maybe Windows (Xinput).

tracernz, 2017-06-30:
Depends Qt 5.9.1 (for deploy support).