d-ronin/dRonin #1561: GCS: Crash on SIGINT

tracernz, 2017-02-07:
* Seen in version (release version or githash): Last month or two, mac and linux

[Cannot figure out why this stacktrace doesn't find symbols on the server](http://dronin-autotown.appspot.com/at/crash/ahFzfmRyb25pbi1hdXRvdG93bnIWCxIJQ3Jhc2hEYXRhGICAgOC_7bQIDA) (they are definitely there/in the correct place) so posting the minidump_stackwalk output when run locally.

Crash is from https://github.com/d-ronin/dRonin/blob/a07db18/ground/gcs/src/plugins/coreplugin/telemetrymonitorwidget.cpp#L197

Looks like this is called from a slot so probably telemetry monitor is in the process of being cleaned up.

```
Operating system: Linux
                  0.0.0 Linux 4.4.0-59-generic #80-Ubuntu SMP Fri Jan 6 17:47:47 UTC 2017 x86_64
CPU: amd64
     family 6 model 60 stepping 3
     1 CPU

GPU: UNKNOWN

Crash reason:  SIGSEGV
Crash address: 0x0
Process uptime: not available

Thread 0 (crashed)
 0  libQt5Core.so.5!QThreadStorageData::get() const + 0x2b
    rax = 0x0000000001d0edd0   rdx = 0x0000000000000020
    rcx = 0x00007fbc9d226730   rbx = 0x0000000001b3b1a0
    rsi = 0x00007fbc9d227050   rdi = 0x0000000000000001
    rbp = 0x0000000000000002   rsp = 0x00007ffdcbb49760
     r8 = 0x0000000000000042    r9 = 0x00000000ffffffff
    r10 = 0x00007fbc9d227050   r11 = 0x0000000000000003
    r12 = 0x0000000000000000   r13 = 0x0000000001b3b1f0
    r14 = 0x00007ffdcbb49820   r15 = 0x00000000050e4018
    rip = 0x00007fbc9b218c1b
    Found by: given as instruction pointer in context
 1  libQt5Gui.so.5!QFontInfo::rawMode() const + 0x3d
    rbx = 0x0000000000000000   rbp = 0x0000000000000002
    rsp = 0x00007ffdcbb497b0   r12 = 0x0000000001d5bf30
    r13 = 0x0000000000000001   r14 = 0x00007ffdcbb49820
    r15 = 0x00000000050e4018   rip = 0x00007fbc9bdde3ad
    Found by: call frame info
 2  libQt5Gui.so.5!QFontPrivate::engineForScript(int) const + 0x44
    rbx = 0x0000000000000000   rbp = 0x0000000000000002
    rsp = 0x00007ffdcbb497d0   r12 = 0x0000000001d5bf30
    r13 = 0x0000000000000001   r14 = 0x00007ffdcbb49820
    r15 = 0x00000000050e4018   rip = 0x00007fbc9bdde504
    Found by: call frame info
 3  libQt5Gui.so.5!QTextEngine::QTextEngine(QString const&, QFont const&) + 0x1a9
    rbx = 0x00000000050e4018   rbp = 0x00007ffdcbb49800
    rsp = 0x00007ffdcbb49800   r12 = 0x0000000003ea0bb0
    r13 = 0x00007ffdcbb49810   r14 = 0x00007ffdcbb49820
    r15 = 0x00000000050e4018   rip = 0x00007fbc9be0b4b9
    Found by: call frame info
 4  libQt5Gui.so.5!QTextLine::layout_helper(int) + 0xb4
    rbx = 0x0000000000000000   rbp = 0x000000007fffffff
    rsp = 0x00007ffdcbb49860   r12 = 0x0000000003ea0bb0
    r13 = 0x00007ffdcbb49e50   r14 = 0x00007ffdcbb49b30
    r15 = 0x00000000050e4018   rip = 0x00007fbc9be23064
    Found by: call frame info
 5  libQt5Gui.so.5!QTextLine::setLineWidth(double) + 0xcb
    rbx = 0x0000000000000000   rbp = 0x0000000003ea0bb0
    rsp = 0x00007ffdcbb49a10   r12 = 0x00007ffdcbb49b50
    r13 = 0x00007ffdcbb49e50   r14 = 0x00007ffdcbb49b30
    r15 = 0x0000000001d158c0   rip = 0x00007fbc9be24cab
    Found by: call frame info
 6  libQt5Gui.so.5!QTextDocumentLayout::resizeInlineObject(QTextInlineObject, int, QTextFormat const&) + 0x53b5
    rbx = 0x00000000ffffff00   rbp = 0x00007ffdcbb49b50
    rsp = 0x00007ffdcbb49a60   r12 = 0x0000000000000001
    r13 = 0x00007ffdcbb49e50   r14 = 0x00007ffdcbb49b30
    r15 = 0x0000000001d158c0   rip = 0x00007fbc9be607f5
    Found by: call frame info
 7  libQt5Gui.so.5!QTextDocumentLayout::resizeInlineObject(QTextInlineObject, int, QTextFormat const&) + 0x64db
[... 848 lines cut ...]
0x7fbc9d0b9000 - 0x7fbc9d0bcfff  385c0604a188198f04d133e54aba7fe7-le64.cache-6  ???
0x7fbc9d0bd000 - 0x7fbc9d0bdfff  dc05db6664285cc2f12bf69c139ae4c3-le64.cache-6  ???
0x7fbc9d0be000 - 0x7fbc9d0c0fff  767a8244fc0220cfb567a839d0392e0b-le64.cache-6  ???
0x7fbc9d0c1000 - 0x7fbc9d0c1fff  4794a0821666d79190d59a36cb4f44b5-le64.cache-6  ???
0x7fbc9d0c2000 - 0x7fbc9d0c6fff  8801497958630a81b71ace7c5f9b32a8-le64.cache-6  ???
0x7fbc9d0c7000 - 0x7fbc9d101fff  365b55f210c0a22e9a19e35191240f32-le64.cache-6  ???
0x7fbc9d102000 - 0x7fbc9d102fff  bab58bb527bb656aaa9f116d68a48d89-le64.cache-6  ???
0x7fbc9d103000 - 0x7fbc9d109fff  3047814df9a2f067bd2d96a2b9c36e5a-le64.cache-6  ???
0x7fbc9d10a000 - 0x7fbc9d111fff  bf3b770c553c462765856025a94f1ce6-le64.cache-6  ???
0x7fbc9d112000 - 0x7fbc9d112fff  56cf4f4769d0f4abc89a4895d7bd3ae1-le64.cache-6  ???
0x7fbc9d113000 - 0x7fbc9d113fff  b9d506c9ac06c20b433354fa67a72993-le64.cache-6  ???
0x7fbc9d114000 - 0x7fbc9d119fff  b47c4e1ecd0709278f4910c18777a504-le64.cache-6  ???
0x7fbc9d11a000 - 0x7fbc9d12cfff  d52a8644073d54c13679302ca1180695-le64.cache-6  ???
0x7fbc9d12d000 - 0x7fbc9d12dfff  551ecf3b0e8b0bca0f25c0944f561853-le64.cache-6  ???
0x7fbc9d12e000 - 0x7fbc9d138fff  d589a48862398ed80a3d6066f4f56f4c-le64.cache-6  ???
0x7fbc9d139000 - 0x7fbc9d139fff  30829fa25452a46451e813d634d7f916-le64.cache-6  ???
0x7fbc9d13a000 - 0x7fbc9d13cfff  75114ca45c98e8a441da0ff356701271-le64.cache-6  ???
0x7fbc9d13d000 - 0x7fbc9d147fff  83bf95040141907cd45bb53cf7c1c148-le64.cache-6  ???
0x7fbc9d148000 - 0x7fbc9d159fff  9b89f8e3dae116d678bbf48e5f21f69b-le64.cache-6  ???
0x7fbc9d15a000 - 0x7fbc9d15cfff  e13b20fdb08344e0e664864cc2ede53d-le64.cache-6  ???
0x7fbc9d19e000 - 0x7fbc9d21dfff  libsystemd.so.0.14.0  ???
0x7fbc9d236000 - 0x7fbc9d236fff  0c9eb80ebd1c36541ebe2852d3bb0c49-le64.cache-6  ???
0x7fbc9d237000 - 0x7fbc9d23afff  7ef2298fde41cc6eeb7af42e48b7d293-le64.cache-6  ???
0x7ffdcbbd4000 - 0x7ffdcbbd5fff  linux-gate.so  ???
```

mlyle, 2017-02-07:
This may be a key reason behind the increased perceived crashiness for me.

tracernz, 2017-02-07:
Found the bug in my symbol collection code... Should only strip the extension before appending ".sym" on Windows, doh (refactored that code recently apparently while suffering a bad case of brainfade). Now I have a whole bunch of symbols to relocate (I did the main ones for this crash by hand last night).

tracernz, 2017-06-17:
This crash moves around to all sorts of widgets if you change code etc. indicating a problem with destruction order. Bisected it down to d0583431cb7075097264c7b70dccfe0f377fea7e which makes a lot of sense. I'll revert that PR and see if I can come up with an alternative (I think I have one).

PR #1823 (2017-06-18): GCS: Replace "SingleInstance" with simple guard
Reverts #1477.

Code from https://stackoverflow.com/questions/5006547/qt-best-practice-for-a-single-instance-app-protection. ~~~The difference to the old impl. (which causes GCS to crash) is that this won't bring the existing instance into focus. I can live without that to avoid segfaults, and perhaps it could be looked at in future.~~~

Fixes #1561

Tested and working on:
- [x] Linux
- [x] macOS
- [x] Windows
closed 2017-06-18 by mlyle