d-ronin/dRonin #1562: Consider asserting/hanging on flash errors / ERRIE

mlyle, 2017-02-07:
As discussed / reported by @mindnever / #1560 --- if we have a null dereference bug we access flash, with undefined behavior.

If it's a write, it just doesn't happen, but also sets the flash error flag.  This causes us to hang on future flash programming operations (e.g. saving a config item).

It would be more helpful to crash immediately, because 1) we'd spot more bugs, and 2) the actual failure/crash would be more proximal to the actual problem code.

Also it's the eeriest flag.

tracernz, 2017-03-23:
Kind of annoying that ST choose 0x0 as the program memory remap. :/ I guess it would actually write (over your code!) if you were running from RAM rather than flash?

mlyle, 2017-03-23:
@tracernz It's ARM's fault, since the initial vector table is at address 0.  A common choice, really, that happens to be super bad for c.  You could make the choice to mask writes to the remapped addresses, though.

And yes, it even risks writing to flash if we've happened to unlock programming, etc.

PR #2121 (2018-03-16): pios_flash: catch flash errors (enable interrupt)
fixes #1562
commit c193a63e (2018-03-16): pios_flash: catch flash errors (enable interrupt)

fixes #1562
closed 2018-03-16 by tracernz