d-ronin/dRonin #1563: Lists on the Authors screen are empty

yamahabest, 2017-02-07:
* Seen in version (release version or githash): Artifice Preview
* On flight controller (if applicable): 

Issue details:
GCS logs the following errors:

21:40:18[WARNING] QFSFileEngine::open: No file name specified
21:40:18[WARNING] QTextBrowser: No document for c:%5CProgram Files (x86)%5CdRonin%5Cshare%5C%5Cgcsrelauthors.html
21:40:18[WARNING] QFSFileEngine::open: No file name specified
21:40:18[WARNING] QTextBrowser: No document for c:%5CProgram Files (x86)%5CdRonin%5Cshare%5C%5Cgcsauthors.html

Files are indeed not in that path, or any other path.

DTFUHF, 2017-02-07:
can reproduce on windows

tracernz, 2017-02-07:
Sigh https://github.com/d-ronin/dRonin/blob/next/package/winx86/dronin.nsi#L168

PR #1566 (2017-02-09): Package/GCS: Fix authors dialog on Win/Linux
Fixes #1563 

Includes whole gcs/share directory in Windows installer (like on other platforms)
closed 2017-02-09 by mlyle