d-ronin/dRonin #1570: No modelviewer on windows

ufoDziner, 2017-02-09:
* Seen in version (release version or githash): Preview-20170207
* On flight controller (if applicable): Sparky2

Issue details:
No model displayed in model viewer on Windows 10.

[image]

tracernz, 2017-02-09:
[image]

tracernz, 2017-02-09:
Background setting was empty for some reason:
[image]

Once set all was well:
[image]

Reset my config (`drgcs -r`), and it worked fine...

ufoDziner, 2017-02-09:
Here are my modelview settings.

[image]

ufoDziner, 2017-02-10:
I tried this on my other windows computer and the model showed up correctly.  I tried the current release, and it's no longer working either.  It has/was working in the past so it appears to be an issue with my video and not the GCS.

mlyle, 2017-02-10:
I just want to reopen this ot see what @tracernz says -- I'm not sure why fully-qualified paths are showing up in this dialog (instead of relative ones within the installation like works).

pug398, 2017-02-10:
Not sure it is related but I see once you extract the working directory dronin-preview-20170207 you cant move it otherwise you break the gcs. Resetting config doesn't fix.

ufoDziner, 2017-02-10:
Here are my GCS settings.  Not sure if it's actually what you're looking for.
[ufo_gcs_settings.zip](https://github.com/d-ronin/dRonin/files/767740/ufo_gcs_settings.zip)

tracernz, 2017-02-10:
Downloaded the preview build (zip), placed on desktop, ran it, closed it, and moved it to another drive, ran it. Some of the QML assets still point to the old path. Some stuff did work though, e.g. github/recent tunes/links on the welcome pane but not the buttons, and PFD is all there except the background image. Modelview was fine.
```
micha@mike-windows MINGW64 /p/tmp/gcs_winx86_amd64-preview-20170207/bin
$ ./drgcs.exe
[Warning] file:///C:/Users/micha/Desktop/gcs_winx86_amd64-preview-20170207/share//welcome/CommunityPanel.qml:8:5: QML BorderImage: Cannot open: file:///C:/Users/micha/Desktop/gcs_winx86_amd64-preview-20170207/share//welcome/images/welcome-news-bg.png
[Warning] file:///C:/Users/micha/Desktop/gcs_winx86_amd64-preview-20170207/share//welcome/WelcomePageButton.qml:27:5: QML Image: Cannot open: file:///C:/Users/micha/Desktop/gcs_winx86_amd64-preview-20170207/share//welcome/images/button-label.png
[Warning] file:///C:/Users/micha/Desktop/gcs_winx86_amd64-preview-20170207/share//welcome/WelcomePageButton.qml:20:5: QML Image: Cannot open: file:///C:/Users/micha/Desktop/gcs_winx86_amd64-preview-20170207/share//welcome/images/wizard-on.png
[Warning] file:///C:/Users/micha/Desktop/gcs_winx86_amd64-preview-20170207/share//welcome/WelcomePageButton.qml:14:5: QML Image: Cannot open: file:///C:/Users/micha/Desktop/gcs_winx86_amd64-preview-20170207/share//welcome/images/wizard-off.png
[Warning] file:///C:/Users/micha/Desktop/gcs_winx86_amd64-preview-20170207/share//welcome/WelcomePageButton.qml:27:5: QML Image: Cannot open: file:///C:/Users/micha/Desktop/gcs_winx86_amd64-preview-20170207/share//welcome/images/button-label.png
[Warning] file:///C:/Users/micha/Desktop/gcs_winx86_amd64-preview-20170207/share//welcome/WelcomePageButton.qml:20:5: QML Image: Cannot open: file:///C:/Users/micha/Desktop/gcs_winx86_amd64-preview-20170207/share//welcome/images/firmware-on.png
[Warning] file:///C:/Users/micha/Desktop/gcs_winx86_amd64-preview-20170207/share//welcome/WelcomePageButton.qml:14:5: QML Image: Cannot open: file:///C:/Users/micha/Desktop/gcs_winx86_amd64-preview-20170207/share//welcome/images/firmware-off.png
[Warning] file:///C:/Users/micha/Desktop/gcs_winx86_amd64-preview-20170207/share//welcome/WelcomePageButton.qml:27:5: QML Image: Cannot open: file:///C:/Users/micha/Desktop/gcs_winx86_amd64-preview-20170207/share//welcome/images/button-label.png
[Warning] file:///C:/Users/micha/Desktop/gcs_winx86_amd64-preview-20170207/share//welcome/WelcomePageButton.qml:20:5: QML Image: Cannot open: file:///C:/Users/micha/Desktop/gcs_winx86_amd64-preview-20170207/share//welcome/images/advanced-on.png
[Warning] file:///C:/Users/micha/Desktop/gcs_winx86_amd64-preview-20170207/share//welcome/WelcomePageButton.qml:14:5: QML Image: Cannot open: file:///C:/Users/micha/Desktop/gcs_winx86_amd64-preview-20170207/share//welcome/images/advanced-off.png
[Warning] file:///C:/Users/micha/Desktop/gcs_winx86_amd64-preview-20170207/share//welcome/WelcomePageButton.qml:27:5: QML Image: Cannot open: file:///C:/Users/micha/Desktop/gcs_winx86_amd64-preview-20170207/share//welcome/images/button-label.png
[Warning] file:///C:/Users/micha/Desktop/gcs_winx86_amd64-preview-20170207/share//welcome/WelcomePageButton.qml:20:5: QML Image: Cannot open: file:///C:/Users/micha/Desktop/gcs_winx86_amd64-preview-20170207/share//welcome/images/scopes-on.png
[Warning] file:///C:/Users/micha/Desktop/gcs_winx86_amd64-preview-20170207/share//welcome/WelcomePageButton.qml:14:5: QML Image: Cannot open: file:///C:/Users/micha/Desktop/gcs_winx86_amd64-preview-20170207/share//welcome/images/scopes-off.png
[Warning] file:///C:/Users/micha/Desktop/gcs_winx86_amd64-preview-20170207/share//welcome/WelcomePageButton.qml:27:5: QML Image: Cannot open: file:///C:/Users/micha/Desktop/gcs_winx86_amd64-preview-20170207/share//welcome/images/button-label.png
[Warning] file:///C:/Users/micha/Desktop/gcs_winx86_amd64-preview-20170207/share//welcome/WelcomePageButton.qml:20:5: QML Image: Cannot open: file:///C:/Users/micha/Desktop/gcs_winx86_amd64-preview-20170207/share//welcome/images/system-on.png
[Warning] file:///C:/Users/micha/Desktop/gcs_winx86_amd64-preview-20170207/share//welcome/WelcomePageButton.qml:14:5: QML Image: Cannot open: file:///C:/Users/micha/Desktop/gcs_winx86_amd64-preview-20170207/share//welcome/images/system-off.png
[Warning] file:///C:/Users/micha/Desktop/gcs_winx86_amd64-preview-20170207/share//welcome/WelcomePageButton.qml:27:5: QML Image: Cannot open: file:///C:/Users/micha/Desktop/gcs_winx86_amd64-preview-20170207/share//welcome/images/button-label.png
[Warning] file:///C:/Users/micha/Desktop/gcs_winx86_amd64-preview-20170207/share//welcome/WelcomePageButton.qml:20:5: QML Image: Cannot open: file:///C:/Users/micha/Desktop/gcs_winx86_amd64-preview-20170207/share//welcome/images/config-on.png
[Warning] file:///C:/Users/micha/Desktop/gcs_winx86_amd64-preview-20170207/share//welcome/WelcomePageButton.qml:14:5: QML Image: Cannot open: file:///C:/Users/micha/Desktop/gcs_winx86_amd64-preview-20170207/share//welcome/images/config-off.png
[Warning] file:///C:/Users/micha/Desktop/gcs_winx86_amd64-preview-20170207/share//welcome/WelcomePageButton.qml:27:5: QML Image: Cannot open: file:///C:/Users/micha/Desktop/gcs_winx86_amd64-preview-20170207/share//welcome/images/button-label.png
[... cut ...]
[Warning] file:///C:/Users/micha/Desktop/gcs_winx86_amd64-preview-20170207/share//welcome/WelcomePageButton.qml:14:5: QML Image: Cannot open: file:///C:/Users/micha/Desktop/gcs_winx86_amd64-preview-20170207/share//welcome/images/flightdata-off.png
[Warning] file:///C:/Users/micha/Desktop/gcs_winx86_amd64-preview-20170207/share//welcome/main.qml:47:13: QML Image: Cannot open: file:///C:/Users/micha/Desktop/gcs_winx86_amd64-preview-20170207/share//welcome/images/welcome-logo.png
[Warning] file:///C:/Users/micha/Desktop/gcs_winx86_amd64-preview-20170207/share/pfd/default/PfdWorldView.qml: The system cannot find the path specified.
```

tracernz, 2017-02-10:
I have a sneaking suspicion it is these qmlc files (compiled/cached QML?).
[image]

tracernz, 2017-02-10:
When I deleted the qmlc files, all became well again.

pug398, 2017-02-11:
That seems to fix the issue I brought up. I can now move the working folder and change the name of the folder it was extracted to and gcs still works proper.

mlyle, 2017-02-27:
Was fixed by #1573

closed 2017-02-27 by mlyle