d-ronin/dRonin #1571: GCS connects only once to powered FC

mluessi, 2017-02-09:
* Seen in version (release version or githash): d93a8de4a660b352366ae303175abaceebbb1e51
* On flight controller (if applicable): Brain, BrainRE1
* OS: Linux

Issue details:
- Power FC from external 5V
- Connect USB, GCS connects to FC normally
- Disconnect USB
- Reconnect USB, GCS does not establish connection

DTFUHF, 2017-02-09:
can repro on Seppuku, Windows, and artifice preview

DTFUHF, 2017-02-09:
Also Linux

Changing computers or restarting GCS doesn't fix the problem, only removing and reapplying power to the FC.

tracernz, 2017-02-11:
I managed to software reboot Seppuku a few times without issue. As soon as I disconnected the USB cable and then reconnected it... it connects and attempts to send session object (GCSTelemetryStats) but looks like nothing is coming back. Flight side issue by the looks? Perhaps uncovered by #1506?

tracernz, 2017-02-11:
This looks like #1384...

tracernz, 2017-02-11:
It's a USB issue. I see it over both HID and VCP, but not over TCP or UART.

tracernz, 2017-02-11:
Quanton is the only F4 board with VBUS connected to the correct pin (PA9) for hardware (USB peripheral) VBUS detection and also the only one that doesn't suffer this bug. ;)

tracernz, 2017-02-11:
Didn't have Brain/Re1 schematics but looking at the hardware defs they should be both PA9 too? They are configured as no pull though. I assume they have external pull-downs?

tracernz, 2017-02-11:
Brain v1 works okay because it uses PA9 like Quanton. Re1 reboots every time I connect USB so can't really test it.

mluessi, 2017-02-11:
RE1 doesn't have an external pull-down on PA9. I can test if enabling the internal one makes a difference.

tracernz, 2017-02-11:
Stopped at a random time while in the bad state: EP1 == HID

OTG_FS_DOEPCTL1, EP1 control register (0x50000B20) = 0x800e8040, or:

31: EPENA: Endpoint enable: 1
30: EPDIS: Endpoint disable: 0
29: SD1PID: Set DATA1 PID: 0
28: SD0PID: Set DATA0 PID: 0
27: SNAK: Set NAK: 0
26: CNAK: Clear NAK: 0
25-22: Reserved: 0
21: STALL: STALL handshake: 0
20: SNPM: Snoop mode: 0
19,18: EPTYP: Endpoint type: 3 = Interrupt
17: NAKSTS: NAK status: 1
16: DPID: Endpoint data PID: 0
15: USBAEP: USB active endpoint: 1
14-11: Reserved: 0
10-0: MPSIZ: Maximum packet size: 64

That looks fine, NAKSTS is only cleared while a transfer is actually in progress.

OTG_FS_DOEPINT1, EP1 interrupt register (0x50000B28) = 0x2010, so we have a random reserved bit (same bit is set on endpoint 0 though), and:
6: B2BSTUP: Back-to-back SETUP packets received: 0
5: Reserved: 0
4: OTEPDIS: OUT token received when endpoint disabled: 1 (but only applies to control endpoints)
3: STUP: SETUP phase done: 0 (but only applies to control endpoints)
2: Reserved: 0
1: EPDISD: Endpoint disabled interrupt: 0
0: XFRC: Transfer completed interrupt: 0

mlyle, 2017-02-13:
Very possibly this is related:

```
root@midnight:/sys/bus/usb/devices# echo 0 > 1-2/bConfigurationValue
root@midnight:/sys/bus/usb/devices# echo 1 > 1-2/bConfigurationValue
```

Crashes the flight controller 100%.

tracernz, 2017-02-13:
This is a dup. of TauLabs/TauLabs#1663. Not a blocker for this release.

mlyle, 2017-02-13:
Yah, sad to defer this one but starting to look necessary.

tracernz, 2017-02-13:
Definitely at least two different bugs. In some cases, DCD_HandleInEP_ISR fires continuously (looks like Tx FIFO empty on endpoint 1), this one would cause a watchdog reboot if I didn't have a debugger connected. In other cases this doesn't happen and LEDs still flash etc. but still no bytes received on HID.

mlyle, 2017-02-13:
Sadly no choice but to defer this.  Would definitely hotfix for this issue, though.

yamahabest, 2017-04-06:
Seems https://github.com/d-ronin/dRonin/issues/1360 was the same issue.