d-ronin/dRonin #1578: Sparky2 WS281x Support

Cro2230, 2017-02-12:
* Seen in version (release version or githash): 
* On flight controller (if applicable): 

Issue details:
Will next release "Artifice" support WS2811 LEDs for Sparky2

yamahabest, 2017-02-12:
Hi, it would be better to ask this on the forum: https://forum.dronin.org/forum/.

I don't think there is support for ws2811 leds on Sparky in the upcoming release.

I would say the following have support right now:
- Seppuku
- BrainRE1
- Quanton
- Revolution

mlyle, 2017-02-12:
I agree this should go to the forum. 

Currently no sparky2 ws2811 support.  Someone would need to step up to implement and/or test it, and what pin is correct for it would need to get figured out.

closed 2017-02-12 by mlyle