d-ronin/dRonin #1602: GCS: Rfm Bind Wizard Appears Broken

tracernz, 2017-02-28:
* Seen in version (release version or githash): Release 20170213
* On flight controller (if applicable): Revo

Issue details:
http://dronin-autotown.appspot.com/at/crash/ahFzfmRyb25pbi1hdXRvdG93bnIWCxIJQ3Jhc2hEYXRhGICAgJDA5JkJDA
http://dronin-autotown.appspot.com/at/crash/ahFzfmRyb25pbi1hdXRvdG93bnIWCxIJQ3Jhc2hEYXRhGICAgJDU59sKDA
http://dronin-autotown.appspot.com/at/crash/ahFzfmRyb25pbi1hdXRvdG93bnIWCxIJQ3Jhc2hEYXRhGICAgJCn4YoLDA
http://dronin-autotown.appspot.com/at/crash/ahFzfmRyb25pbi1hdXRvdG93bnIWCxIJQ3Jhc2hEYXRhGICAgJDt4IELDA
http://dronin-autotown.appspot.com/at/crash/ahFzfmRyb25pbi1hdXRvdG93bnIWCxIJQ3Jhc2hEYXRhGICAgJDAgMgIDA
http://dronin-autotown.appspot.com/at/crash/ahFzfmRyb25pbi1hdXRvdG93bnIWCxIJQ3Jhc2hEYXRhGICAgJDJk54LDA
http://dronin-autotown.appspot.com/at/crash/ahFzfmRyb25pbi1hdXRvdG93bnIWCxIJQ3Jhc2hEYXRhGICAgNDAypsJDA

mlyle, 2017-02-28:
Yuck.  OK.

tracernz, 2017-02-28:
Keep in mind those are all the same guy. Needs triage. :smiley:

tracernz, 2017-03-08:
He managed to get this to work somehow. Would still be good to understand the failure but this can probably be punted from the hotfix so we can get RE1 fixed.

utoedter, 2017-03-31:
I can confirm that crash too, happens for me on Windows 7/64 Bit. If you need more information let me know.

pug398, 2017-04-09:
If you get the time could you revisit the RF Bind wizard. It works for me in 78241e0d 20170317
but appears broken in 1a2c64cc 20170330 and later. The latter is the first build I have after 20170317 but does not mean that is where problem started.

tracernz, 2017-04-30:
With a freshly booted PipX, it will not reply to an HwTauLink request. If you send the object from GCS it will work after that. The bind wizard relies on getting a reply to the HwTauLink request to identify the board. Need to look into firmware to set why it doesn't work anymore.

tracernz, 2017-04-30:
@pug398 https://github.com/d-ronin/dRonin/pull/1758 should fix your issue.

pug398, 2017-04-30:
@tracernz both taulink and oplink modules are operational again. Excellent work! Thanks :)

mlyle, 2017-06-02:
Fixed by #1758

closed 2017-06-02 by mlyle