d-ronin/dRonin #1603: Artifice Hotfix 1

tracernz, 2017-02-28:
Issue to track/discuss hotfix (@mluessi, @mlyle).

To be fixed:
- [X] RE1: RGB LEDs broken #1587 
- [ ] Authors list on Authors screen still empty #1600
- [ ] GCS: Rfm Bind Wizard Appears Broken #1602

mlyle, 2017-03-09:
@mluessi --- artifacts are going to spit out soon and I'll upload them to https://github.com/d-ronin/dRonin/releases/tag/Release-20170213.1 within the next couple of hours.  Can you please verify LEDs on RE1 and overall function with RE1 ASAP?  (It's marked pre-release but I'll push it out more generally once you OK.)

Also, your assistance with getting #1523 out of limbo would be appreciated...

mlyle, 2017-03-09:
@mluessi -- artifacts are up-- will you be able to look at this today?

mluessi, 2017-03-09:
@mlyle thanks. I'm travelling at the moment without access to hardware to test things. I can test it this weekend.

mlyle, 2017-03-10:
Shipped

closed 2017-03-10 by mlyle