d-ronin/dRonin #1612: Stabilization: Failsafe provisions don't work properly after apparent code refactor.

glowtape, 2017-03-05:
* Seen in version (release version or githash): Artifice

The Stabilization module is in case of a failsafe situation supposed to pull a multirotor into a level position and stop any yaw movement, and make a plane to spiral down slowly. It does this by overriding the flight modes and seting a specific attitude and yaw rate. Apparently there was some refactoring that shuffled things around, essentially bypassing this.

The attitude error calculation that controls the leveling happens before these failsafe provisions. Furthermore the rate loop considers a different input variable than what these provisions override.

The flight modes get overridden as expected, but the desired stabilisation still gets applied, and potentially misinterpreted in case of switching blindly from rate to attitude mode.

The code in question here:
https://github.com/d-ronin/dRonin/blob/next/flight/Modules/Stabilization/stabilization.c#L496

local_attitude_error calculation happens above that.

The rate loop below uses stabDesired[i] instead of raw_input.

glowtape, 2017-03-05:
Submitted #1613 as proposal. Simple code movement.

Right now, as asked in IRC and desired by @mlyle, axes with contuining control input remain functional.

However, _usually_ RPY+Thrust usually come from the same receiver, and when it goes into failsafe, all axes are gone. If we make the assumption that there aren't any significant numbers of unicorn configurations with two or more receivers to control the four main inputs, we could simplify the code a bit to make it more readable.

I suppose one could program the failsafe to put certain axes into failsafe manually (via invalid values) while put valid ones into other axes, but I don't think that should be encouraged.

mlyle, 2017-03-07:
fixed by #1613

closed 2017-03-07 by mlyle