d-ronin/dRonin #1622: Quanton: Serial Tx Broken

tracernz, 2017-03-10:
* Seen in version (release version or githash): Release-20170213.1 and next
* On flight controller (if applicable): Quanton

Issue details:
Flight controller doesn't send any bytes back on Tx pin when UART1 configured for Telemetry at 115200 or "Init HM10" (aka 115200) (didn't try any others, went to Seppuku instead which worked okay).

mlyle, 2017-03-10:
Dumb question-- you set UART4 which defaults to Telemetry on Quanton to Disabled, right?

mlyle, 2017-03-10:
Hmmm.. works for me (once I got rx/tx onto the right sides :P

tracernz, 2017-03-10:
> Dumb question-- you set UART4 which defaults to Telemetry on Quanton to Disabled, right?

Ahh... WTF... that's probably it.

closed 2017-03-11 by mlyle