d-ronin/dRonin #1623: Moving Horizontal Stripes on OSD layout RE1

RoadyFPV, 2017-03-10:
Issue details:
Hi i have flashd latest dronin on my BreinFPV re1. Here i have n OSD issue, moving Horizontal Stripes from up to down on OSD layout. With the vido sync detector i cant fix it. 
when i flash the older firmware dRonin 2016-10-04.1 Release ("Quixote"). The OSD is fine.

mlyle, 2017-03-10:
Do you have any screenshots so we can better understand?

Are you running dshot?

mlyle, 2017-03-10:
(Have many users on RE1 on Artifice with no reports of this issue, so I suspect it's something unique to your configuration...)

RoadyFPV, 2017-03-10:
I running D-Shot600.
screenshot coming tomorrow.
Thx for fast help

mlyle, 2017-03-10:
dshot support is preliminary and I wouldn't really recommend it for OSD targets.  It's almost certainly your problem.

closed 2017-03-10 by mlyle