d-ronin/dRonin #1629: Fix QT version stuff over at readme.io

glowtape, 2017-03-12:
Both v20170213 and v999-dev still refer to QT 5.6.1 and MinGW 4.9.2. Not that there's people running down our door, writing code, but @yamahabest is currently having a ball with that one.

yamahabest, 2017-03-12:
Yeah, thanks.
And while we're at it, I can't seem to solve the following compile error:

dRonin\ground\gcs\src\libs\libcrashreporter-qt\libcrashreporter-handler\Handler.cpp(33): fatal error C1083: Cannot open include file: 'client/windows/handler/exception_handler.h': No such file or directory.

The file seems to be there on that path, but the compiler can't find it, it seems.

tracernz, 2017-03-13:
I've been working on making Windows environment setup easy (Qt SDK download and install is taken care of like on Linux and Mac) and not breaking old branches every time we upgrade a tool. Current progress is at https://github.com/tracernz/dronin/tree/mrc-win-ng (CAUTION: don't clone this, I'll be force-pushing to the branch). I think it's pretty close.

tracernz, 2017-03-13:
Remaining work is to check for a valid MSVS installation before installing Qt etc. (if setting up an MSVC environment), and maybe look into MS Build Tools as an alternative to installing MSVS.

mlyle, 2017-07-18:
superseded d-ronin/d-ronin.github.io#47

closed 2017-07-18 by mlyle