d-ronin/dRonin #1631: androidgcs crashes when selecting the Tuning view

mkroehnert, 2017-03-13:
* Seen in androidgcs version Artifice on a Moto G3 running Android 6.0.1

Issue details:

* open androidgcs
* open left drawer
* tap on `Tuning`
* androidgcs crashes independent of being connected to a flight controller or not

The crashlog can be downloaded [here](https://drive.google.com/open?id=0B-EyIKkOzbhyaW1pdWFFeHBBeGc). Unfortunately GitHub didn't let me upload the log or a zipped version of it to this issue.

mlyle, 2017-03-23:
I can't repro this on my Nexus 5X on android 7.1.1 or my ancient tablet running 4.4.  I was on 6.0 for awhile and had no issues with this either :/

mlyle, 2017-03-24:
@mkroehnert Theory-- could this be a locale issue?  Are you in a locale that uses , for a decimal separator?  I know androidgcs crashes when something that isn't 0-9 or . is put into a tuning field.

mkroehnert, 2017-03-24:
@mlyle thanks for looking into this issue. It could be a locale issue, since my phone uses German localization.

mkroehnert, 2017-03-24:
@mlyle let me know if I can do some beta testing before a new version gets released.

mlyle, 2017-03-24:
@mkroehnert I will try to look at this soon.  Thanks for the report.   I have some other things I'm working on that will need to take priority though.

mlyle, 2017-03-24:
@mkroehnert OK i got a chance sooner than I thought I would ;)

http://jar.lyle.org/~mlyle/agcs.apk   Note you'll need to remove the one from the store before installing this, as it's unsigned.

closed 2017-03-25 by tracernz

mkroehnert, 2017-03-25:
@mlyle sure, no problem. This is also not a high-priority issue on my side.

mkroehnert, 2017-03-25:
@mlyle thanks for the fast response. I can confirm that androidgcs does not crash anymore with the version provided by you. 

Next week I will check what happens when I connect to a flightcontroller.

mkroehnert, 2017-03-30:
I can confirm that the androidgcs version provided in this PR works also, when attached to a powered flight controller.

mlyle, 2017-03-30:
@mkroehnert excellent, thank you for the confirmation :D

PR #1658 (2017-03-25): androidgcs: changes to number parsing
Don't crash when a letter shows up, do locale-specific parsing, blah
blah.  The way this is written is still **not good**.  But this is
minimal fix.

Fixes #1631

Would be **much** cleaner with Java 8 features like default methods in interfaces.
commit 6986d00b (2017-03-25): androidgcs: changes to number parsing

Don't crash when a letter shows up, do locale-specific parsing, blah
blah.  The way this is written is still **not good**.  But this is
minimal fix.

Fixes #1631