d-ronin/dRonin #1641: I-term accumulator gets reset every loop in stabilization.c

glowtape, 2017-03-17:
I only noticed this while debugging my predictor code, because values kept getting reset, that shouldn't be (because said function was only supposed to be called when an axis mode changes). Then I started tracking number of calls to the function and noticed it went sky high and kept rising no matter what.

Turns out the previous_mode array in stab.c, used to determine whether a mode on an axis was switched, gets reset to 255,255,255 every loop. The piece of code responsible is this one, because when I comment out the for-loop, the issue stops:

https://github.com/d-ronin/dRonin/blob/next/flight/Modules/Stabilization/stabilization.c#L993

From what I can tell looking at UAVOs, the FlightStatus.Armed field used in the if-statement never gets set to Armed when the quad arms (and works, since it flies). Tracking it down, it seems like something related to the control events stuff between manualcontrol.c and transmitter_control.c that was eventually introduced.

Anyway, the problem is, due to this, reinit always evaluates to true and keeps resetting the I-term accumulators. This is a big problem.

mlyle, 2017-03-17:
??

```
(gdb) b stabilization.c:360
Breakpoint 1 at 0x56572308: file /home/mlyle/dRonin/flight/Modules/Stabilization/stabilization.c, line 360.
(gdb) cont
Continuing.
Watchdog fired / 39377170

Breakpoint 1, stabilizationTask (parameters=0x0)
    at /home/mlyle/dRonin/flight/Modules/Stabilization/stabilization.c:362
362			float dT = PIOS_DELAY_DiffuS(timeval) * 1.0e-6f;
(gdb) p previous_mode
$1 = "\002\002\001"
(gdb) print *pids[0]
Structure has no component named operator*.
(gdb) print pids[0]
$2 = {p = 0.00200000009, i = 0.00150000001, d = 0, iLim = 0.300000012, 
  iAccumulator = 0.000214389278, lastErr = 0.541875422, lastDer = 0}
(gdb) print flightStatus
$3 = {Armed = 2 '\002', FlightMode = 6 '\006', ControlSource = 2 '\002'}
```

glowtape, 2017-03-17:
On the quad itself, FlightStatus.Armed stays Disarmed, for whatever reason whatsoever. Any way I can derby this on the controller without special hardware?

??

mlyle, 2017-03-17:
So when I hook my quad up to GCS, and flip the arming switch, I see a little gear ("arming") and then the checkbox ("armed").  Those correspond to flightstatus values 1 and 2.

mlyle, 2017-03-17:
No chance that arming got set to "always disarmed" or something is there?  Or a regression in your branch?

 i term definitely does things.. even undesired things (had a fun windup-related transient on takeoff during tests yesterday) -- in flight in current code.

mlyle, 2017-03-17:
(also note that if I is disabled almost all quadrotors will spin freely in yaw).

glowtape, 2017-03-17:
It's my regular quad configuration that always worked (in some fashion anyway). The branch is basically Artifice with just my Crossfire+Telemetry code and the predictor stuff that only touches the new raw_input fields. I haven't touched anything in regards to arming, manual control or related. 
When I'm back home later today, I'm gonna configure SBus n.i. to exclude the RX and check stock Artifice by dumping its counts of reinit evaluating to true into some unused UAVO.
This is really odd. Actually, I've always considered my quad a little slower than expected on things like flips and such. Noticed that again after some LoS antics yesterday (what was supposed to be 720deg/s looked like maybe half of it). The I-term missing would explain it. Why it doesn't considerably yaw on its own despite that, I don't really know (maybe my motors aren't worn enough so that the P-term is sufficient?).

mlyle, 2017-03-17:
Does motors spin while armed work?  It also uses flightstatus.armed.

Does gyro-zeroing at arming work?  It also uses flightstatus.armed...

P term is never sufficient to remove a bias...

I strongly suspect a methodology problem...

glowtape, 2017-03-17:
User error. Battery monitoring created a critical alarm, so benchtesting don't work :[

closed 2017-03-17 by glowtape