d-ronin/dRonin #1674: No such signal Core::ConnectionManager::deviceConnected

mlyle, 2017-03-27:
```QObject::connect: No such signal Core::ConnectionManager::deviceConnected(QIODevice *) in /home/mlyle/dRonin/ground/gcs/src/plugins/uavtalk/uavtalkplugin.cpp:65
QObject::connect: No such signal Core::ConnectionManager::deviceAboutToDisconnect() in /home/mlyle/dRonin/ground/gcs/src/plugins/uavtalk/uavtalkplugin.cpp:67
[Warning] QObject::connect: No such signal Core::ConnectionManager::deviceAboutToDisconnect() in /home/mlyle/dRonin/ground/gcs/src/plugins/scope/scopegadgetwidget.cpp:88
[Warning] QObject::connect: No such signal Core::ConnectionManager::deviceConnected(QIODevice*) in /home/mlyle/dRonin/ground/gcs/src/plugins/scope/scopegadgetwidget.cpp:89
[Warning] QObject::connect: No such signal Core::ConnectionManager::deviceAboutToDisconnect() in /home/mlyle/dRonin/ground/gcs/src/plugins/scope/scopegadgetwidget.cpp:88
[Warning] QObject::connect: No such signal Core::ConnectionManager::deviceConnected(QIODevice*) in /home/mlyle/dRonin/ground/gcs/src/plugins/scope/scopegadgetwidget.cpp:89
[Warning] QObject::connect: No such signal Core::ConnectionManager::deviceAboutToDisconnect() in /home/mlyle/dRonin/ground/gcs/src/plugins/scope/scopegadgetwidget.cpp:88
[Warning] QObject::connect: No such signal Core::ConnectionManager::deviceConnected(QIODevice*) in /home/mlyle/dRonin/ground/gcs/src/plugins/scope/scopegadgetwidget.cpp:89
[Warning] QObject::connect: No such signal Core::ConnectionManager::deviceAboutToDisconnect() in /home/mlyle/dRonin/ground/gcs/src/plugins/scope/scopegadgetwidget.cpp:88
[Warning] QObject::connect: No such signal Core::ConnectionManager::deviceConnected(QIODevice*) in /home/mlyle/dRonin/ground/gcs/src/plugins/scope/scopegadgetwidget.cpp:89
```

tracernz, 2017-03-27:
Oops. QtCreator didn't see those. :( Scope gadget should definitely be connected to TelemetryManager instead!

mlyle, 2017-03-27:
I think this prevents connection to the FC.  trying revert to make sure we're clean/OK

PR #1675 (2017-03-27): Revert #1670 "GCS:Wizard: Correct connection dropdown and button state"
This reverts commit 9708603c65671c186d12245b779521ff723be977.

Fixes #1674
commit ef10535b (2017-03-27): Revert #1670 "GCS:Wizard: Correct connection dropdown and button state"

This reverts commit 9708603c65671c186d12245b779521ff723be977.

Fixes #1674
closed 2017-03-27 by mlyle