d-ronin/dRonin #1678: GCS: User inteface design discussion (RFC)

glowtape, 2017-03-29:
(I'm gonna close this issue once everything's discussed/clarified.)

There's a bunch of things I'd like an yay, nay and/or opinion on, when it comes to overhauling/streamlining the UI eventually:

- I'd like to enforce a color scheme, likely light/web-like (what it ends up being I'm not sure yet, that'll happen organically while trying things). When trying to apply some more graphical design elements (re: color coding) to an UI, there comes a point you can't rely on the system color scheme to fit every time. Considering some users having funny ideas about system UI colors.

- I think one customization options I'd love to see inaccessible is the user's ability to change SVGs. How many people even use or need this? If that's an OK thing to do, we could then consider redirecting all SVG loads to a specific folder instead of default, deluxe, hi-contrast* and whatever, which would allow us to enforce the new SVGs in weird configs and scrub the resources down the line, too.

Before I'm even going to start with any of this, I need to spend some time with QT, to learn QML and see how practical it is to replace the .ui stuff with QML. The UI markup files get loaded into widgets and its objects are specifically addressed through a single object instance (they're not integrated into the class via a preprocessor, like say what happens in XAML for instance), so I'm hoping it's possible to drop in QML files without changing much code.  But I'm feeling like there's a big catch coming. Ignoring the can of worms of redoing the UI stuff to begin with.

(*: There's these SVGs, but I have a hard time finding some sort of accessibility option in the GCS. Given the color scheme of the new SVGs, we could do away with these, too?)

mlyle, 2017-03-29:
> I'd like to enforce a color scheme, likely light/web-like (what it ends up being I'm not sure yet, that'll happen organically while trying things). When trying to apply some more graphical design [...]

Yah, I'm mostly OK with this.  It may not be able to be **strictly** uniform though-- different platforms may still want different colors; do you think we can work with this?  (I want to avoid us becoming too un-native looking if possible)

> I think one customization options I'd love to see inaccessible is the user's ability to change SVGs

Yes, kill this IMO.

> Before I'm even going to start with any of this, I need to spend some time with QT, to learn QML and see how practical it is to replace the .ui stuff with QML. 

I am not a fan of this except in specific cases.   Probably most importantly, I'm skeptical of rewrites-- they're a great chance to introduce regression and we have a **lot** of UI surface.  Also even though QML is really neat it's not strictly better-- a whole lot more is runtime-evaluated and both can be expected to be slower and to not have various kinds of bugs caught at compile time.

I am not positive, but QML seems like it introduces opportunity for things to behave differently than what QT Creator creates for UI files, too.

If a widget is going to be shared with mobile, or does lots of "webby" stuff, etc, QML seems like a natural fit.  It may be worth doing what you're talking about for one or two classes that otherwise need a lot of help just to get experience with it and learn the caveats.  But at this time I'm not in favor of wholesale-replacing "stuff that works".

tracernz, 2017-03-29:
> I think one customization options I'd love to see inaccessible is the user's ability to change SVGs. 

This is SUPER handy for development as you don't need to rebuild GCS to see changes. What do we gain by removing it?

glowtape, 2017-03-29:
QML is out for the big stuff then. Can live with that. The idea behind this was, if some parts of the UI are subject to large rearranging, maybe just rewrite it in the more modern stuff (which I understand what QML is). Alas, skipping the QMLing is probably less work for me after all.

As far as the colors go, all three supported operating systems feature some shade of light grey by default, and it'd be nice to enforce these default colors, so that the UI still looks "right" whenever there's big quasi-white surfaces inside. On the other hand, it probably doesn't make a difference for most people. But I'd love to prevent someone coming in and complaining that the UI looks like crap, just to find out that said person thinks that a pink or dark blue UI looks baller and the GCS looks like it has clown make-up on.

About the SVG icons, some things like the dials currently have multiple subdirectories for different versions of their icons and the default configurations pick from at least two of these. It'd be nice, if we could hide it and rewrite the path from existing installations to pick from a single directory. This runs on the assumption that rarely anyone goes to customize the icons (I didn't even know you could do that until I touched the default configurations). This'd let us do some cleaning up. I suppose a compromise would be to leave things as they are in the debug builds and hide it and rewrite the paths in the release builds?

tracernz, 2017-03-29:
Again, what is the advantage of removing the configuration functionality? Removing all but one configuration I'm fine with, but seems silly to pull out the code which hasn't required any maintenance and doesn't look like it will anytime soon. Removing the possibility will make life harder for developers (I use multiple configurations regularly).

The only difference between debug and release builds should be additional checks (including debug versions of upstream libraries). Not any functional difference.

glowtape, 2017-03-29:
To clarify, I don't want to scrap the configurations themselves. I want this to be away from the user:

[image]

tracernz, 2017-03-29:
Yeah I want that because I use it. I don't see that removing it will help anyone, but it will be a pain for me.

tracernz, 2017-03-29:
This kind of configuribility and the generality of it is one of the things I like most about GCS..

glowtape, 2017-03-29:
Hmmm. OK, here's maybe an alternative that might work and let us keep that field while allowing to clean up the SVGs: Adding some code in the section that loads the SVGs in the respective widgets, that checks whether a file exists, and if not, it'll rewrite the path to look into specific subdirectory (say "dronin" instead of default, deluxe or whatever). This wouldn't break the dials and whatever when a future releases omits certain resource folders. That would work?

(Not now, anyway, at some point later.)

tracernz, 2017-03-29:
Can you explain what you think the issue is and how it manifests? Deleting the resource folder will break existing configurations whether the path pointing to it is fixed or configured through the XML config.

glowtape, 2017-03-29:
So, with the updated Flight Data screen, I've updated some SVGs. But they're updated in place. I've updated them in place because it's probably not a wrong assumption that most users run a default configuration, and ideally I'd like them to see the updated dials, so that the "new" Flight Data screen looks kinda consistent.

However, they're spread across the default/ and deluxe/ folders*, because that's how the four configuration templates refer to them. I'd love to see things in a single folder, to allow to tidy up things a bit (it appears there's plenty of unused graphics spread over share/ anyway). However if we were to consolidate everything to a single folder, this'd break existing configurations. New installations would be taken care of by updating the configuration templates. However, once there's been a first run, it doesn't matter what the templates say.

So the issue is how and whether to deal with folders going missing. When I look at my current dr.xml, the current SVGs in question are referred to like this:

`%%DATAPATH%%dials/deluxe/barometer.svg`

If that path doesn't exist, because we ditched the deluxe folder, you could rewrite it to:

`%%DATAPATH%%dials/dronin/barometer.svg`

...and the affected widget would still be happy because it finds an SVG.

Of course, this also hinges on whether we want or need two sets of dials, default and "deluxe" ones. I'm not sure why, but I'm open to reasoning. If it's latter, all becomes moot and we have to decide how to keep the two existing sets _and_ make the new dials show up where desired.

*: Given that the new Flight Data screen is a little darker than before, it could be argued whether we still need the high contrast versions (third folder) or not. I'd love to hear an opinion about this.

(I'm not sure how to keep this shorter, because I feel I'm unable to make my point. I've already rewritten this thrice :D )

tracernz, 2017-03-29:
Right. I would suggest we just delete whichever set are not the defaults and don't worry about what happens to users not running defaults.

glowtape, 2017-03-29:
Ditching the redundant SVGs is definitely good. Trying to consolidate the containing folders is probably too much of a bother in retrospect.

So, in regards to colors and configurability thereof (as per comments in #1669)...

I can change things back so that the Flight Data screen fully respects the User Interface Color in the options screen. It'd just need changing some things to transparent in the SVGs for them to work with it, and deal specifically with the GPS signal dial (the way the bars were handled doesn't work with transparent). However I'd be interested for running installations to change it to the current new color the first run of the new release, so I have to figure out a solution for this. If an user choses to still change it (back) after that, so be it.

Re: system colors. I can work around that, blindly assuming that users either keep the defaults (being the aforementioned light grey shades all three OSes use) or deviate from that sensibly.

mlyle, 2017-04-07:
Any further discussion needed here?

glowtape, 2017-04-07:
Nope. Gotta spend some time trying a bunch of things practically first, so there's not much to discuss for now.

closed 2017-04-07 by glowtape