d-ronin/dRonin #1679: VCP Oddities

mlyle, 2017-03-29:
* [ ] If you read with a long timeout (e.g. PIOS_QUEUE_TIMEOUT_MAX), the port locks up for at least Linux users.
* [ ] The port is unusable for several seconds after connection when using Liunx
* [ ] VCP to BLHeliSuite doesn't seem to work (cabled serial does).

All experienced on F4 on Seppuku

tracernz, 2017-03-29:
Hmmm, I think I was doing the 4w-if stuff over VCP with BLHelisuite. Although I do tend to use a hardware serial for Dev work because USB ports dissappear when the firmware stops so can't be sure.

mlyle, 2017-03-29:
I get "SetCommState function failed"   With Windows Message : "System Error.  Code: 31.A device attached to the system is not functioning" 

Probably some nuances of VCP implementation that need work..

tracernz, 2017-03-29:
I ran under WINE which probably explains why it would work for me. :wink:

mlyle, 2017-03-29:
Looks like when opening with blhelisuite, URB requests 33 (GET_LINE_CODING) is sent twice unanswered pertaining to the IN endpoint, and 32 (SET_LINE_CODING) is sent once, and SET_CONTROL_LINE_STATE is sent once.

mlyle, 2017-03-30:
[image]

PR #1780 (2017-05-23): 4WIF: Change 2: Fixes to USB CDC implementation
1. Just send a sane line state report always; don't try and omit it and
confuse the other side (wtf!)
2. Don't inhibit TX based on "serial lines" like DTE.
3. Simplify and make tx_active tracking logic in CDC layer correct.

edit: fixes #1679
closed 2017-05-23 by tracernz