d-ronin/dRonin #1746: Wrong implementation of axis-lock

fotis400, 2017-04-23:
* Seen in version (release version or githash): 
Possibly all, tested with firmware #190852f3.

* On flight controller (if applicable): 
OpenPilot Revolution

Issue details:
The current implementation of axis lock is wrong. It is not true axis-lock, it is unusable for helis, where it is very necessary.

I am currently setting up a Trex 450 Pro (helicopter), with an OpenPilot Revolution board. For helicopters the axis lock functionality is VERY important for the yaw axis. The current implementation seemed to me that is not correct by reading the docs, but I also tried it in real life and it got confirmed. Excessive and unpredictable tail drift, not flyable IMO.

Note that it is normal for this not to have been observed in the past, as symmetrical airframes (as for example the popular quadcopters), are not affected by this that much. Asymmetrical airframes however, like a helicopter and possibly a tricopter suffer a lot.

 The whole concept has to change.

### What the firmware does now:
The yaw axis is set to axis lock. While flying and with zero stick input, the tail remains locked in place, as it ought to. By minimal input (more than `MaxAxisLockRate` ), the tail unlocks and enters rate mode. The rate of yaw follows the stick input (or at least in theory).

### What is the problem:
By minimal input the tail "unlocks". Since the PID needs some error to work on, and since the helicopter by its nature has the tendency to yaw, drifting of the tail starts. This drift rate is of course constantly changing, as it depends on any cyclic or collective inputs. In fact without the axis-lock functionality it is normal for all helis in rate mode to experience excessive yaw rate. Furthermore due to this phenomenon the current implementation causes significant different yaw rates on the left and right side.

Here is what actually happened to me, and I consider the most annoying/dangerous. Lets say for example that when the tail is unlocked (just in rate mode) it has a tendency to drift at 20deg/s to the left. With no stick input the tail is locked, and the performance acceptable. If now I set my stick to 2deg/s to the right, then although I requested for slight *right* yaw, the tail suddenly unlocks from  causing an 18deg/s yaw to the *left*! Such small yaw adjustments, are not only very needed, but also can happen accidentally in fast collective movements.

### How it has to be implemented:
**The tail has to be locked at all times!** Unlocking the tail for any reason, is a recipe for disaster. Also forget about rate. Rate mode is *completely* unrelated to axis-lock. In fact axis lock is very similar to attitude mode. The difference is that in attitude mode the accelerometers are used to determine the "level", while in axis-lock a *virtual level* has to be used. This virtual level has to be free to rotate, and must be manipulated by the transmitter sticks. To further explain it, it has to be as the heli is always set to self-level, but somehow I have magically coupled my transmitter sticks to `BoardRotation`. I instruct the virtual level to rotate, and the heli follows trying to self-level again.

#### Behavior Example:
The heli sits in the air locked. I instruct with my yaw stick a rate of turn of 100deg/s to the right, for 1 second. This means that the new heading has to be 100 degrees to the right. The heli, no matter what happens (e.g. wind, collective/cyclic input etc), has to finally be found heading 100 degrees to the right. The actual rate this will happen is indifferent. The faster the better. The initial 100deg/s command was related to the rotation of the heading, not the rate of turn of the heli itself!

The above are crucial for heli stabilization. It will help all other unsymmetrical airframes. But apart from yaw (and helis), it could be used to all axis, in all airframes, leading in VERY tight control, and almost zero influence of any external factor (e.g. wind) and any airframe tendencies. It will also lead in very consistent control regardless of the battery/motors state and performance.

mlyle, 2017-04-23:
I agree our axis lock implementation is not good, but I don't agree completely with this.

It's also bad to just integrate a rate command and come up with desired attitude.   Because if the axis underperforms for whatever reason, it may "catch up" to the command later and be a big surprise.

Note that there's been quite a few people flying on helicopters with this exact concept of axis lock across multiple projects implementing basically this logic.

> Lets say for example that when the tail is unlocked (just in rate mode) it has a tendency to drift at 20deg/s to the left. 

Hm.  That sounds **super** bad if the rate loop is completely ineffective.  Do you have sufficient inner loop integral gain (or do you need a higher ilimit term)?  Even rate mode should be able to mostly hold a constant heading.

fotis400, 2017-04-24:
Well I understand your points, but this are only theoretical concerns, not actual.

> It's also bad to just integrate a rate command and come up with desired attitude. Because if the axis underperforms for whatever reason, it may "catch up" to the command later and be a big surprise.

Yes this is normal. This is how it has been for the last 20 years or so in the helicopter world. And this is how it is implemented in the "high-end" flybarless controllers. You will see warnings about this all over the place, in all manuals and forums. Pilots are aware of this.

On the one hand this is the desired behavior, at least in helicopters. If I want it to stay to a specific heading, *it has to*. The purpose of the gyro is to catch-up for any kind of underperformance, either by the tail rotor, motor bogging, wind etc. During these incidents, as the tail control is degraded, the PID will briefly be detuned. The axis-lock has to correct it.

On the other hand such under performance exist only in theory. Helis do perform as they should. We never experience such underperformance. Never. Don't forget that a heli has a MUCH more yaw authority than a quad. It is very safe in my opinion, and there should be no worries on this topic. Currently there is a max deflection that axis-lock tries to correct. This safety feature may also be kept, so a surprise will be of only 30 degrees, as opposed to a complete pirouette. But also note that many flybarless controllers do not even have this. The heli will pirouette as many complete turns as needed to compensate.

To have a surpise, the underperformance has to be very severe. If for example I request for a 360deg turn, and the heli is completely unable to follow, but it does so after a second for example, then this heli is very very problematic. Even in extreme cases these surprises will be in the scale of just a few degrees.

> Hm. That sounds super bad if the rate loop is completely ineffective. Do you have sufficient inner loop integral gain (or do you need a higher ilimit term)? Even rate mode should be able to mostly hold a constant heading.

Nope, my PID seems quite close to the optimal. Keep in mind that the axis-lock is there to catch these deflections that a simple PID controller cannot. Read bellow.

> Even rate mode should be able to mostly hold a constant heading.

Generally it is almost impossible to fly a heli, with just a rate gyro. Many years ago, when we only had rate gyros, we used the so called **revolution mix** in our transmitters. Later transmitter manufacturers added other related features too. The purpose of a revo mix, is to apply a constant yaw correction depending on the collective input. So you had to create a curve matching the heli's tendency to yaw for all flying conditions, and collective commands, to yaw adjust in advance, not leaving room for the gyro to drift. Obviously the rate of turn was closed-loop, but the actual attitude was open-loop, resulting in very tedious tuning, and poor performance. This it the problem heading hold gyros solved.

So rate in yaw means, revo mix. Without it, it is in the physics to have drift and poor control.

fotis400, 2017-04-24:
Oh and another thing. In attitude mode the flight computer decides the fastest path to the desired attitude. Requesting for self-level, while rolled right, will result in a left roll, etc.

In axis-lock, the direction of the stick input has to be maintained. For example if I request a super fast 340 degree right pirouette, the heli may not have have the time to perform the maneuver yet. It has to use the actual initial requested turn to the right (340degs), instead of choosing the shortest path to the left (20 degs).

mlyle, 2017-04-24:
> To have a surpise, the underperformance has to be very severe. If for example I request for a 360deg turn, and the heli is completely unable to follow, but it does so after a second for example, [...]

There are large sets of flight regimes of aircraft where an axis becomes saturated and completely unresponsive.  Fewer on helicopters, but definitely on multirotors and fixed wing.  It is really bad if rudder gets stuck right and there's not anything a pilot can do about it.

> Generally it is almost impossible to fly a heli, with just a rate gyro. 

Hmm.. I flew a helicopter with just a rate gyro (P) a decade ago; and then built a controller that did what was effectively PI control of yaw; so I'm not completely without knowledge in this area.   Note that a PI controller does **a lot more** than rate gyro products which are generally just P controllers.  A PI controller hunting for 0 with any kind of bandwidth should be able to hold an axis stationary, and the whole purpose of the axis lock is to just fix minor perturbations.

@ekaszubski --- how is yaw control?  Do you have trouble with the tail coming unlocked and tail drift?

fotis400, 2017-04-24:
Well yes definitely, I also have used rate gyros for helis. But again only with a revo mix, and surely is by far inferior to an axis lock system. The PI is improvement against the P only controller, but only regarding the rate of turn. Not the aircrafts actual attitude.

I was thinking however your concerns today. I came up with the following idea, that it will solve any such issues.

The implementation changes to my proposition. Thus the axis-locked stick input has to be integrated (to determine the target attitude). Simultaneously the rate of turn is monitored. By the time the rate PID error gets larger than X, input integration stops. The control continues with best-effort rate mode, till it catches up again with the expected.

This gives axis-lock, with no possibility of windup, and will cause the manoeuvre to halt by the time the stick is centered. 

Another simple option is just to respect 0 stick position. So the axis remains in axis lock mode for zero stick (exactly as it is now), but any other stick input is treated as attitude control, instead of rate. This will also give the pilot the ability to quickly halt the rotation.

mlyle, 2017-04-24:
> The implementation changes to my proposition. Thus the axis-locked stick input has to be integrated (to determine the target attitude). Simultaneously the rate of turn is monitored. By the time the [...]

> This gives axis-lock, with no possibility of windup, and will cause the manoeuvre to halt by the time the stick is centered.

> Another simple option is just to respect 0 stick position. So the axis remains in axis lock mode for zero stick (exactly as it is now), but any other stick input is treated as attitude control, instead of rate. This will also give the pilot the ability to quickly halt the rotation.

You've basically described how the loiter controller works--  if you move the stick while in a loiter/position hold mode, the position setpoint is integrated, but it also decays back towards the current position to prevent it from getting unsafely ahead.  I've planned to do this for a variant of axis lock, but it's not a very high priority.

closed 2017-04-24 by mlyle

ekaszubski, 2017-04-29:
Sorry to show up late, but yaw control has been great for all my helis, 325mm - 515mm.