d-ronin/dRonin #1763: Check: Hex-H motor mixing.

mlyle, 2017-05-06:
I'm not mr. check for correct signs, but it sure seemed like it was wrong before...

jhitesma, 2017-05-08:
Not sure what you mean by "Hex-H"  I only see "Hexa" and "Hexa-X" - though X does look kind of like an H in our drawing.  I've been flying the X on my hex since day one without any issues.  Well, other than having to remember that our motor rotations are reversed from what *flight used so I always have to click the reverse all motors box since this build originally had *flight on it since I can't be bothered to cut off the heat shrink and switch the rotation jumpers on the ESC's.

LitterBugs, 2017-05-25:
I'm working on a F550 Hex with a DTFc and can give it a check.

LitterBugs, 2017-06-02:
Tested on Flamewheel F550 Hex with DTFc.  No issues flying.

mlyle, 2017-06-02:
Nice, thanks.  I must have had map wrong or something.

closed 2017-06-02 by mlyle