d-ronin/dRonin #1767: angry build output on OSX (even on successful builds)

mlyle, 2017-05-09:
```
 CC        [fw|bre1]  flight/Modules/VtolPathFollower/vtol_follower_fsm.c
 Reading /Users/Shared/Jenkins/Library/Developer/org.jenkins-ci.slave.jnlp/workspace/dronin/Nodes/osx/ground/gcs/src/crashreporterapp/crashreporterapp.pro [/Users/Shared/Jenkins/Library/Developer/org.jenkins-ci.slave.jnlp/workspace/dronin/Nodes/osx/build/ground/gcs/src/crashreporterapp]
error: unknown option `contains'
usage: git for-each-ref [<options>] [<pattern>]

    -s, --shell           quote placeholders suitably for shells
    -p, --perl            quote placeholders suitably for perl
    --python              quote placeholders suitably for python
    --tcl                 quote placeholders suitably for Tcl

    --count <n>           show only <n> matched refs
    --format <format>     format to use for the output
    --sort <key>          field name to sort on

error: unknown option `points-at'
usage: git for-each-ref [<options>] [<pattern>]

    -s, --shell           quote placeholders suitably for shells
    -p, --perl            quote placeholders suitably for perl
    --python              quote placeholders suitably for python
    --tcl                 quote placeholders suitably for Tcl

    --count <n>           show only <n> matched refs
    --format <format>     format to use for the output
    --sort <key>          field name to sort on
```

mlyle, 2017-05-12:
Issue is the old git version that comes with mac os x.  I've done brew install git to get a newer version, but for whatever reason the jenkins slave does not prefer it like the baseline path on the user it runs under.

This is likely why things are marked dirty, too, right?

tracernz, 2017-06-03:
> This is likely why things are marked dirty, too, right?
Quite likely. Introduced in Git 2.7 by the looks.

tracernz, 2018-02-09:
Really want to fix this for wired. Stops the tag from being attached to crash reports.

tracernz, 2018-02-09:
What version of macOS are you running on the builder? I've got git 2.14.3 (Apple Git-98) on my machine and it doesn't suffer this.

PR #2057 (2018-02-10): GCS:crashreporter: Fix git version info
- Hopefully make it work with the ancient git version on the mac builder, whilst sticking with plumbing commands
- Force mainwindow.cpp to recompile (unfortunately always for now)
    - Previously the version info only changed if crashreporterapp was rebuilt for some other reason

Fixes #1767
closed 2018-02-10 by mlyle