d-ronin/dRonin #1799: GCS: Telemetry status bar overlaps tab controls

tracernz, 2017-06-03:
Regression from #1669. The main problem is it shadows almost all of the click area for the scroll button, and also it covers a lesser amount visually.
[image]

Would be nice if we dropped that box altogether when space is really tight I think.

tracernz, 2017-06-23:
I could fix the visual issue but not the mouse click area overlap, deferring.

glowtape, 2018-01-09:
I shifted the tab-bar a couple of pixels to the right for visual reasons. It renders it as such, but doesn't adapt the layouting to the shift. I spend a short moment trying to fix it while keeping the visual, but apparently the tab-bar doesn't give a shit about the box model. Gonna give it another go later on.

tracernz, 2018-01-10:
Remember you're operating within the QWidget layout model. W3C CSS box model doesn't apply, and QSS box model I imagine is subject to constraints of the widget layout.