d-ronin/dRonin #1807: RC Configuration Wizard settings fail to save with a timeout

garrinkimmell, 2017-06-07:
* Seen in version (release version or githash): Release-20170213.1: 20170309-190852f3
* On flight controller (if applicable): Release-20170213.1   (git hash 190852f3) (Seppuku FC)

Issue details:

I ran the GCS RC Input "Configuration Wizard" to completion, but the settings failed to save at the end. The portion of the system log indicating the failure appears to be the following. Full log is at [1].  

```
22:45:01[debug] [smartsavebutton.cpp] Sending object to remote end -  "ManualControlSettings"
22:45:01[debug] [smartsavebutton.cpp] Save request for object "ManualControlSettings"
22:45:03[debug] [smartsavebutton.cpp] Saving timeout for object "ManualControlSettings"
22:45:03[debug] [smartsavebutton.cpp] failed to save: "ManualControlSettings"
22:45:03[debug] [smartsavebutton.cpp] Error, tried to apply/save a non-settings object:  "ManualControlCommand"
22:45:03[debug] RESULT "Objects not saved:\nManualControlSettings\n" () ("ManualControlSettings") ()
```

Additionally, I switched to the `System` tab and clicked the `Save` button. This didn't produce any log output, but the modified values for my transmitter setup remained bolded (which AFAICT indicate that they are different than the saved settings on the board?), and the settings did not survive a reboot

I did find a workaround (though clumsy) where I `Export UAV Settings...` [2], reboot, then `Import UAV Settings...`, then save in the `System tab`. After that, the RC input settings seem to persist through subsequent reboots.

[1] https://gist.github.com/garrinkimmell/f0a81e3847c399ec2bd180bcad5ef8b4
[2] https://gist.github.com/garrinkimmell/762a52df46e4a8e616189a20fcdf9815

mlyle, 2017-06-07:
@garrinkimmell --- I'm not sure what you're encountering as I've not seen it before.  Since you have a config dump already, can you try resetting the configuration partition to see if things change?

Go into the firmware uploader, click 'rescue', plug in seppuku.  Right click on "settings" in the partition browser at right, and select "Erase" from the menu.  After booting and importing your config, are you able to save?

The critical line in the log is "Saving timeout for object "ManualControlSettings".  The next one is a dumb bug that creates angry error output and prevents a green checkmark from appearing on the Save button, but does no other harm.

garrinkimmell, 2017-06-07:
@mlyle I had actually done a settings partition erase immediately before running the config wizard -- you've been helpful to me in a variety of forums lately :) As it is now, I _am_ able to import the config and that config is persisted across reboots. When I get a chance tonight I'll try another partition erase followed by an import of that config and let you know. 

One thing I did notice was that if I changed a setting in the `System` tab UAVO browser, then hit the red `Save` button, I didn't see any output in the log indicating that a save had been successful (or that it had failed). Is that the expected behavior? I can always power cycle the board and see if the setting persisted, but I didn't know if there is another indication I should be seeing in the UI.

mlyle, 2017-06-07:
Strange.

Do you have any unusual modules running or unusual configuration?

One oddity we've encountered in the past, is that if someone tries to write to a NULL pointer, it puts the STM32's flash controller into an error state and we can't program config successfully.  I'm not aware of any bugs like this, but if you have something unusual turned on (mavlink? etc) it could be a factor.

garrinkimmell, 2017-06-08:
Just tried the following:

1. Erase settings partition, reboot, load configuration in gist above, saved (on `System` tab), reboot. Settings were persisted.

2. Manually edited the configuration file to set Uart3 to "Disabled" from "GPS"  (the only module checked in the `Modules` tab), saved, and exported configuration. 

3. Re-ran the steps in #1 above, except using the sans-GPS configuration from step 2 above. Ran the Radio Setup Wizard, went to save when completed, original failure occurred.

I'm new to this, so there may be something obvious in the config that's unusual, but it's getting by me. I do really appreciate you having taken the time to try to diagnose it with me.

mlyle, 2017-06-08:
@garrinkimmell Can you please attach your configuration or email to mlyle@lyle.org to see if i can save it to a seppuku and then make further changes?

mlyle, 2017-06-08:
Investigated-- unfortunately this was #1657

closed 2017-06-08 by mlyle