d-ronin/dRonin #1812: Radio Configuration Wizard Accessory2 taking input from Accessory1

yds, 2017-06-13:
* Seen in version (release version or githash):  908e5eed
* On flight controller (if applicable):

Issue details:  When going through the Radio Configuration Wizard I get the following correct mapping:

Accessory0: Chan 7
Accessory1: Chan 9
Accessory2: Chan 10

However the on-screen representation toggles **both** _Accessory1_ and _Accessory2_ when the _Chan 9_ switch is toggled and does not toggle anything when _Chan 10_ gets toggled.

To be clear: this is only an issue with how the toggle verification is represented on screen.  The actual mapping after the wizard completes is correct.

yamahabest, 2017-06-13:
https://github.com/d-ronin/dRonin/issues/1272

mlyle, 2017-06-13:
Dup #1272, thx

closed 2017-06-15 by mlyle