d-ronin/dRonin #1830: Test macOS 10.13 High Sierra beta

tracernz, 2017-06-20:
Before we release, make sure there are no show-stoppers.

mlyle, 2017-06-28:
@tracernz -- are you a developer program member?  I am not, and am not sure it's worth $99 to test this.  Public beta is in "late June"

tracernz, 2017-06-28:
Yup, had trouble installing it on my VM though. :( Might upgrade my laptop.

tracernz, 2017-07-02:
Tested GCS with TCP and HID telemetry on High Sierra developer beta 2. A couple of minor graphical issues but nothing bad (and possibly they aren't specific to High Sierra anyway). Will keep this open for now to test with whatever is the latest release right before we release.

tracernz, 2017-07-02:
Xcode 8.3.3 complains about sim build a bit, but it does still work.
```
 AR        [fw|sim ]  build/sim/firmware.a
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib: file: /Users/mike/dev/dronin/build/sim/firmware.a(betaflightshim.o) has no symbols
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib: file: /Users/mike/dev/dronin/build/sim/firmware.a(betaflightshim2.o) has no symbols
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib: file: /Users/mike/dev/dronin/build/sim/firmware.a(betaflightshim3.o) has no symbols
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib: file: /Users/mike/dev/dronin/build/sim/firmware.a(rgbleds.o) has no symbols
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib: file: /Users/mike/dev/dronin/build/sim/firmware.a(pios_bmm150.o) has no symbols
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib: file: /Users/mike/dev/dronin/build/sim/firmware.a(pios_bmp280.o) has no symbols
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib: file: /Users/mike/dev/dronin/build/sim/firmware.a(pios_bmx055.o) has no symbols
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib: file: /Users/mike/dev/dronin/build/sim/firmware.a(pios_flyingpio.o) has no symbols
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib: file: /Users/mike/dev/dronin/build/sim/firmware.a(pios_hmc5883.o) has no symbols
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib: file: /Users/mike/dev/dronin/build/sim/firmware.a(pios_hmc5983.o) has no symbols
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib: file: /Users/mike/dev/dronin/build/sim/firmware.a(pios_i2c.o) has no symbols
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib: file: /Users/mike/dev/dronin/build/sim/firmware.a(pios_ms5611.o) has no symbols
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib: file: /Users/mike/dev/dronin/build/sim/firmware.a(pios_ms5611_spi.o) has no symbols
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib: file: /Users/mike/dev/dronin/build/sim/firmware.a(pios_px4flow.o) has no symbols
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib: file: /Users/mike/dev/dronin/build/sim/firmware.a(pios_omnip.o) has no symbols
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib: file: /Users/mike/dev/dronin/build/sim/firmware.a(pios_serial.o) has no symbols
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib: file: /Users/mike/dev/dronin/build/sim/firmware.a(pios_spi.o) has no symbols
```

tracernz, 2017-07-13:
Preview build runs okay on macOS 10.13 beta 3. Restoring my laptop to 10.12 because beta 2 and 3 were too unstable for normal use so can't test anymore.

closed 2017-07-13 by tracernz