d-ronin/dRonin #1853: GCS does not start on 10.12.5

mlyle, 2017-06-27:
Seen since last night's nightly.

mlyle, 2017-06-27:
```
Process:               dRonin-GCS [33727]
Path:                  /Applications/dRonin GCS.app/Contents/MacOS/dRonin-GCS
Identifier:            org.dronin.gcs
Version:               1.3.1 (1.3.1)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           dRonin-GCS [33727]
User ID:               501

Date/Time:             2017-06-27 15:55:03.315 -0700
OS Version:            Mac OS X 10.12.5 (16F73)
Report Version:        12
Anonymous UUID:        1841267F-3853-6436-EC77-9C79EC169DFE

Sleep/Wake UUID:       4ADEF72A-DC7B-4E99-A340-837AEE317942

Time Awake Since Boot: 190000 seconds
Time Since Wake:       670 seconds

System Integrity Protection: enabled

Crashed Thread:        0

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    DYLD, [0x1] Library missing

Application Specific Information:
dyld: launch, loading dependent libraries

Dyld Error Message:
  Library not loaded: @rpath/QtGui.framework/Versions/5/QtGui
  Referenced from: /Applications/dRonin GCS.app/Contents/MacOS/dRonin-GCS
  Reason: image not found

Binary Images:
       0x1045ee000 -        0x104616ff7 +org.dronin.gcs (1.3.1 - 1.3.1) <A6890230-012E-3331-97E9-4DD0DE10566D> /Applications/dRonin GCS.app/Contents/MacOS/dRonin-GCS
       0x104630000 -        0x10464eff7 +libUtils.1.dylib (0) <B41A443A-125D-3869-B21E-FCD7955D9596> /Applications/dRonin GCS.app/Contents/Plugins/libUtils.1.dylib
       0x10894b000 -        0x108988dc7  dyld (433.5) <322C06B7-8878-311D-888C-C8FD2CA96FF3> /usr/lib/dyld

Model: iMac15,1, BootROM IM151.0207.B16, 4 processors, Intel Core i5, 3.5 GHz, 8 GB, SMC 2.22f16
Graphics: AMD Radeon R9 M290X, AMD Radeon R9 M290X, PCIe, 2048 MB
Memory Module: BANK 0/DIMM0, 4 GB, DDR3, 1600 MHz, 0x80AD, 0x484D54343531533641465238412D50422020
Memory Module: BANK 1/DIMM0, 4 GB, DDR3, 1600 MHz, 0x80AD, 0x484D54343531533641465238412D50422020
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x142), Broadcom BCM43xx 1.0 (7.21.171.126.1a2)
Bluetooth: Version 5.0.4f18, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en1
Serial ATA Device: APPLE SSD SD0128F, 121.33 GB
Serial ATA Device: APPLE HDD ST1000DM003, 1 TB
USB Device: USB 3.0 Bus
USB Device: BRCM20702 Hub
USB Device: Bluetooth USB Host Controller
USB Device: FaceTime HD Camera (Built-in)
USB Device: MT1956
Thunderbolt Bus: iMac, Apple Inc., 26.1
```

mlyle, 2017-06-27:
```
sharpie:Plugins mlyle$ otool -l libUtils.dylib  | less
...
Load command 12
          cmd LC_LOAD_DYLIB
      cmdsize 64
         name @rpath/QtGui.framework/Versions/5/QtGui (offset 24)
   time stamp 2 Wed Dec 31 16:00:02 1969
      current version 5.8.0
compatibility version 5.8.0
...
Load command 25
          cmd LC_RPATH
      cmdsize 136
         path /Users/Shared/Jenkins/Library/Developer/org.jenkins-ci.slave.jnlp/workspace/dronin/Nodes/osx/tools/Qt5.8.0/5.8/clang_64/lib (offset 12)
Load command 26
```

tracernz, 2017-06-28:
https://github.com/d-ronin/dRonin/blob/b323ecce4bc16b80aa638fdff07793a872d2e3f7/ground/gcs/src/rpath.pri#L2

tracernz, 2017-06-28:
Actually plugins get it from here... https://github.com/d-ronin/dRonin/blob/b323ecce4bc16b80aa638fdff07793a872d2e3f7/ground/gcs/src/gcsplugin.pri#L44

mlyle, 2017-06-28:
Last time this #1331 was mysteriously cured, perhaps relating to a mac os x builder reboot...

so uh, I'm going to do that..

mlyle, 2017-06-28:
Rebooted it, fixed ... suspect this is really #1341.

closed 2017-06-28 by mlyle