d-ronin/dRonin #1854: Failure to perform Automatic Level Detection after Temperature Compensation

nanodude, 2017-06-28:
* Seen in version (release version or githash): Release-20170213.1
* On flight controller (if applicable): BrainFPV RE1

Issue details:

After performing Temperature Compensation, it is no longer possible to perform an Automatic Level Detection. Looking at the Sensor Settings on the browser on the RHS, a few of the values are NANs. After these NANs are changed to zero, Automatic Level Detection works again.

mlyle, 2017-06-28:
For now I suggest you avoid temperature compensation

nanodude, 2017-06-28:
Thank you, Michael. The NANs don't appear on any of the polynomial coefficients, though. As far as I can tell, the coefs are legit after a 45C temp ramp. If I remember correctly, the NANs appeared on AccelBias, AccelScale and ZAccelOffset, but not all components.

tracernz, 2017-06-28:
I tried without changing the temp, then was able to do one automatic level detection, then a second attempt caused the firmware to crash. Attitude solution was obviously going crazy the whole time due to temperature noise + bogus coeffs. Simple solution seems to be just disable the "Accept Temperature Calibration" button until we've actually seen the temperature change by at least "Temperature range"?

mlyle, 2017-06-28:
The temp calibration is pretty .. finnicky from what I remember, and it's easy to get values in the polynomial that result in crazy oscillation etc.  I think it really needs a thorough examination.

But it's not really worth the effort, because even multirotor nav works pretty well without gyro tempcal?  :shrug:

nanodude, 2017-06-28:
Thanks for your work, guys!  The funny thing is that tempcal worked perfectly, with very reasonable coefs (essentially a straight line for the three axes). For some reason, tempcal wrote some garbage values into AccelBias, AccelScale and ZAccelOffset even though I was under the impression that only the gyro stuff was affected.

mlyle, 2017-06-28:
@nanodude-- is the ongoing calibration we do with the accelKI term ineffective?  it has always worked OK for me.

nanodude, 2017-06-29:
@mlyle I forget exactly which variables where affected. I'm going to repeat the calibration tonight, time permitting, and will post the results. Thanks!