d-ronin/dRonin #1866: libjasper1 not available for Ubuntu 17.04 and higher

auturgy, 2017-06-30:
* Seen in version (release version or githash): all
* On flight controller (if applicable): GCS

Issue details: libjasper1 is a dependancy for DroninGCS in linux.  This library is not available via standard archives on newer releases, as it is not a maintained package. It can be downloaded from here http://launchpadlibrarian.net/245417065/libjasper1_1.900.1-debian1-2.4ubuntu1_amd64.deb and installed via dpkg, which works on Ubuntu 17.04, but is essentially untested.

mlyle, 2017-07-01:
Hey, sorry-- unfortunately this is an upstream issue with Qt---

```
mlyle@midnight:~/dRonin/tools/Qt5.8.0/5.8/gcc_64/plugins/imageformats$ ldd libqjp2.so | grep -i jasp
	libjasper.so.1 => /usr/lib/x86_64-linux-gnu/libjasper.so.1 (0x00007f957d1be000)
```

Whenever they fix it, we'll pick it up.  It looks like Debian unilaterally deleted the package because of lack of a maintainer a couple of months ago.

closed 2017-07-01 by mlyle

auturgy, 2017-07-01:
No worries: I'm guessing not many DRonin users are on Linux. Might be worth putting a note in your installation docs though.

Regards,

James

tracernz, 2017-07-01:
Couldn't find any mention of this in Qt Jira, so likely they don't know about it. I see it in [Zesty Zapus (17.04)](https://launchpad.net/ubuntu/zesty/+package/libjasper1) but not in [Artful Aardvark (17.10)](https://launchpad.net/ubuntu/artful/+package/libjasper1)?

tracernz, 2017-07-01:
Looks like it's still maintained? https://github.com/mdadams/jasper/commits/master

auturgy, 2017-07-01:
Might be one of those silly examples where the source is maintained but no one wants to be the package manager (at least for Debian based distros).

Regards,

James

mlyle, 2017-07-01:
@tracenrz https://launchpad.net/ubuntu/zesty/amd64/libjasper1/1.900.1-debian1-2.4+deb8u1

status: deleted :(

tracernz, 2017-07-02:
@auturgy GCS tends to work best on Linux because it gets the most dev time (and I think that's true of upstream Qt too). :)

auturgy, 2017-07-02:
Cool.
I've only just started looking at dronin - been using Ardupilot recently but found a brainfpv in an old racer I thought which I've put in a new frame, and taulabs seems to have gone stale (James Cotton has gone very quiet in general actually).
I look forward to seeing where you guys keep taking things - much more sophisticated than beta flight, not quite as complex as Ardupilot. An interesting place to be in.

Regards,

James