d-ronin/dRonin #1871: After upgrade/uavo import, before reset: UI panes consider themselves unsaved

mlyle, 2017-07-01:
You're supposed to reboot anyways-- but if you don't, you get wonky results.

tracernz, 2017-07-02:
The wizard should reboot at the end IMO, that would also result in automatically bringing up the setup wizard if you don't have actuator settings (although the upgrader still doesn't handle no settings properly either so that won't be perfect).

mlyle, 2017-07-02:
Yes, but caution needs to be used here-- I think that reboot could race with commits if done naively currently

tracernz, 2017-07-02:
Yeah, would be safe with a 5 second timer or something?

tracernz, 2018-01-01:
I prepared a fix for this (https://github.com/tracernz/dronin/commit/df0f7635510082259fbe42a2b1c3ac2f255c2297)... but I can't actually repro this with an import or an upgrade on current next (be3307d). Can you please try next, and if you can still repro, try that branch?

mlyle, 2018-03-17:
I have a recent repro clicking through input stuff ---- will try to repro and check out your fix.

mlyle, 2018-03-17:
Actually it's output stuff.

mlyle, 2018-03-17:
I can still repro with the fix.  Procedure: import a UAV file that changes actuatorsettings from defaults->nondefaults or vice versa.  (Importing the same one doesn't trigger it!)  Then click to output pane, then somewhere else.

mlyle, 2018-04-19:
So, to note: the output pane is one of the few that doesn't auto-build relations.

mlyle, 2018-05-09:
Another one: stabilization acts "unsaved" on connection of pipxtreme.  My guess, the backing objects are missing and it calculates new values and seems unsaved.