d-ronin/dRonin #1872: Neat test issue

mlyle, 2017-07-01:
Rolling up results from light test.

mlyle, 2017-07-01:
* Lux / 180 lasered frame: lost mixer during upgrade(but was on weird intermediate build).  Default PIDs wobbled; autotune worked despite constant wobble
* original brain on blade 150: worked great
* TBS Vendetta (Lux/Colibri Race): flight was great. '2811 LEDs worked-- but blinked 'I' or similar during flight for undetermined reasons.  Could not repro on bench.
* Seppuku small quad: all great.  Tested stacked tune, and that was pretty good too.
* Seppuku HUGE quad: all great.  Sounds terrifying :P 
* RE1 Owl-style: Required change to sensorsettings-- cutoff=48, filterorder=3 to tune-- had TONS of vibration before that / bumpy autotune part.  Flies great after tunem but the actual aircraft is falling apart.
* Navigation plus quad: all good.

mlyle, 2017-07-05:
Relnotes in progress:

* Core flight firmware improvements
  * "NeoTune" next generation cross-correlation based tune algorithm yields better results-- even in the presence of vibration and other anomalies.
  * Improved sensor filtering implemented in the Sensor module.
  * Improvements to attitude estimation performance in the presence of vibration.
  * Support 4WIF passthrough for ESC programming across 'MSP' protocol interface.
  * Improvements to realtime behavior and removal of jitter.
  * Mixer architecture changes improve performance and allow custom mixers that use "accessory" data across outputs (and lays the groundwork for future control improvements).
  * Several fixes made to telemetry/configuration protocol performance across flight and GCS.
* Changes in supported targets
  * Support added for PikoBLX micro flight controller
  * Support added for Seriously Pro F3 EVO ("SPRF3E") and various clones (e.g. FrSky XSRF3E).  Note that only the "EVO" SPRF3 variants are supported.
  * REMOVED SUPPORT FOR F1 FLIGHT CONTROLLERS (CC3D, NAZE).  Use "Artifice" for these controllers. 
* Target improvements
  * Support for WS2811/2812 LEDs added to select F3 targets (Lux, SPRF3E)
  * DShot using DMA implemented for OSD targets (original Brain, Seppuku, and BrainRE1)
  * Improvements to on screen display reliability on original Brain and Seppuku flight controllers.
  * Various improvements to the AQ32 target: flash erase, DAC annunciation, and WS2811 features.  However, PWM RX support was removed.
  * Moved all Invensense targets to a common, improved sensor driver.
  * Simulation/flyingpi: move to OS-native threading model instead of embedding ChibiOS; put the simulator binaries for ARM and x86 in the Linux packages.
* External hardware support
  * Support for TBS Crossfire receiver protocol
  * Preliminary support for OmniPreSense millimeter-wave radars.
  * Improvements to IBus protocol reliability.
* Miscellaneous
  * Simple UI added for annunciator (LED, buzzer) settings.
  * Various bugfixes and performance improvements.
  * Improvements to UI styling.

tracernz, 2017-07-13:
Preview build runs okay on macOS 10.13 beta 3. Restoring my laptop to 10.12 because beta 2 and 3 were too unstable for normal use so can't test anymore.

closed 2017-07-17 by mlyle