d-ronin/dRonin #1876: GCS/Autotown: Record tune sample data in Autotown

tracernz, 2017-07-04:
Would be nice to grab at very least the raw tune data (a few kb binary blob) and stick it in autotown as a blob (cloud storage, linked to the datastore entries we already do?). @dustin

tracernz, 2017-07-04:
@dustin I wonder if `handleStoredTune` could return some kind of key in body and GCS could then send a PUT request with the big blob to /storeTuneBlob/<secretkey>. storeTuneBlob would store the blob in cloud storage, then decode the key and attach the blob to the tune record already in datastore?

mlyle, 2017-07-04:
Is it unacceptable to just put the tune data base64 encoded in a json field?  would be about 44kbyte

tracernz, 2017-07-05:
It seems undesirable to mix binary and JSON data, and especially to make it bigger by base64 encoding. Whatever is easiest for @dustin works for me though.

mlyle, 2017-07-05:
Yah, it's yucky.. but it involves no new work and no new indirection, and the total size isn't something completely ridiculous-- datastore is 18 cents per gigabyte * month, so about 100 user tunes for a penny per year.  (Blob storage is 1/10th the price, but our time is worth something).

PR #1877 (2017-07-09): Autotune: Send compressed raw tune data to cloud
Also display per-axis taus properly.

Fixes #1876
closed 2017-07-09 by dustin