d-ronin/dRonin #1879: Flight: Reclaim Init Task Stack

tracernz, 2017-07-09:
Memory is a bit tight on some targets. We could gain back 1 kiB by locating the init task stack adjacent to the end of the heap (would do this in linker script rather than mallocing it at runtime) and adding it to the heap when the init task deletes itself.

mlyle, 2017-07-10:
System task could be taught to start the rest of the modules tasks--- which would also have the effect of providing stronger ordering so we don't do config checks prematurely etc.

tracernz, 2017-07-10:
The only problem I see with that is that System is no longer a module, but an integral part of PiOS..

mlyle, 2017-07-10:
Yah, it's slightly unfortunate.  OTOH, main.c (and module start) just got factored out of targets in Neat.. having system call something to complete startup doesn't seem **too** unclean.

System task also does stuff like checks the malloc failed stuff during startup to do the boot fault things, and connects to object persistence-- which really should have a defined order with all the rest of "stuff" (otherwise some of these things could go off in the wrong order, e.g. someone tries to save something before System is listening).

tracernz, 2017-07-10:
It's not all that different to CorePlugin in GCS :stuck_out_tongue: (although the PluginManager does load and start all the plugins).

glowtape, 2018-01-09:
Any particular reason why we shouldn't just do it in main() and skip initTask? IIRC, from what I have seen with the ChibiOS patch, because we allow an idle thread, main() gets turned into a thread after chSysInit(). There's 1KB stack reserved for the main thread, anyway.

tracernz, 2018-01-10:
There would be no way to get that memory back? Afaik the os requires that thread to always exist. This only seems important on F3 (only 40 k) and PipX (20 k).

glowtape, 2018-01-10:
Hm, what do you mean get memory back? The process stack that gets used for the main thread is statically allocated in the linker script currently. So it's gone anyway.

The patch referred above skips the initTask thread and runs the stuff directly on main. My Seppuku quad fires up battery monitoring, logging, altitude hold, MSP bridge and Crossfire telemetry next to the necessary flight modules. The lowest I've seen unused process stack being reported after a bunch of reboots was 408 bytes, seems there's plenty of headroom for annoying module inits. We don't really need to spawn a separate thread for initTask.

We could do the heap expansion thing as you suggested, though, stick the process stack at the end of RAM, and stick a check in PIOS_malloc whether the main thread is suspended and expand. ChibiOS 17 might throw a spanner into this, tho. We're using ChibiOS' mempool in pios_queue, which now requires memcore in the new version. I'd have to see whether it can easily be easily replaced with a construct of our own, so that we can keep using simple alloc.

glowtape, 2018-01-10:
Hmmm, if we move the idle loop into main(), too, we could get rid of the idle thread and save that thread working area. Gotta test it when I'm up later today.

tracernz, 2018-01-10:
> Hm, what do you mean get memory back? The process stack that gets used for the main thread is statically allocated in the linker script currently. So it's gone anyway.

Init task needs more than 1 k of stack for at least some of the modules IIRC. So you'd have to make the process stack bigger, with no hope of being able to free that memory and add it to the heap after module init/start is done.

glowtape, 2018-01-10:
Do you happen to remember which modules? initTask only allocates 1KB currently.

PR #2122 (2018-04-13): Run system task in init task (to save task stack space)
Fix #1879.

Saves a bunch of RAM.  Requires analysis of system task priority under this new scheme.  (Both posix & not).  Other small fixes.

Needs test on hardware (tested sim with and without realtime).
closed 2018-04-13 by tracernz