d-ronin/dRonin #1911: GCS: Wizard can't choose port Win10

rlesac, 2017-08-08:
* Seen in version (release version or githash):  Release-20170717: 20170715-1283f272
* On flight controller (if applicable): SR F3

Issue details: 
Upon starting the wizard, all the available ports are grayed out and can't be chosen.

[image]

mlyle, 2017-08-08:
We should probably filter out serial from here--- only USB attached (autoconnected) flight controllers are supported by the wizard, I think.

tracernz, 2017-08-08:
https://github.com/TauLabs/TauLabs/issues/1861 would be nice, may look at it one day.

mlyle, 2017-08-10:
@tracernz -- flyingpi/sim would benefit.

PR #2124 (2018-03-18): setupwizard: don't show anything except USB devices
(Because nothing else makes sense-- setup wizard over radio etc is silly, and all normal targets have USB, and you can always pre-connect the device you want if you are really the rare exception).

Fixes #1911
closed 2018-03-18 by tracernz