d-ronin/dRonin #1940: Boot loop on SPR F3 EVO if mag is set to disabled

LitterBugs, 2017-09-07:
* Seen in version (Neat Release-20170717): 
* On flight controller (Sprf3e): 

Issue details:
I have verified this on two different FCs.  Originally found it with an EMAX Femto SPRF3E clone that does not have an internal mag.  Was trying to get rid of the sensor error that was being thrown for the missing mag.  As soon as I disabled the mag and rebooted, the board would go into a boot loop.  Had to clear the settings partition to get the boot loops to stop.  Tried several times, and every time the mag is disabled, the board will get stuck in a boot loop.
Also verified it with a genuine SPR F3 Evo board.  Board was working fine with mag enabled.  As soon as mag is disabled, it gets into the same boot loop and can only be recovered by clearing the settings partition.

LitterBugs, 2017-09-07:
Acutally, it has the Magnetometer, it is missing the Baro.  But if either is disabled, it will get stuck in a boot loop.

tracernz, 2017-09-09:
Can confirm, 2x flashes repeated 3 times == gyro (http://dronin.org/docs/initial-setup/blink-beep-codes/)... I wonder if we should make these beeps happen for a bit longer so it's more distinguishable from a watchdog @mlyle?

glowtape, 2018-01-09:
My proper SPRF3e doesn't complain with disabled mag. If @LitterBugs verifies it working on his variant, I guess this could be closed to make @mlyle happy. Issue count and what not. :P

LitterBugs, 2018-01-09:
I believe this was fixed a while back.  OK to close.

closed 2018-01-09 by mlyle