d-ronin/dRonin #1948: flight: *_t is reserved by POSIX

tracernz, 2017-09-21:
It has been used a bit recently in flight code. We should probably avoid it to keep away from potential clashes in future.

Relates #1947

mlyle, 2017-09-21:
:/ this is one of those things that's technically true, but every project and coding standard violates--  I just grepped a dozen source trees including Linux, GCC, flightgear, Eigen, etc, and found it used as a preferred suffix for type names in each  Can't we live with it?

tracernz, 2017-09-21:
I think we can live with it but should avoid it without pios_ prefix? IOW <relatively generic name>_t is a recipe for problems, but pios_<whatever>_t is quite safe.

mlyle, 2018-03-18:
What do you think, add to style guide to generally avoid-- pios_whatever_t or in libraries subsystem_whatever_t ... and close this out?  (And add no new violators)

glowtape, 2018-03-18:
I like subsystem_whatever_t.

tracernz, 2018-03-18:
Why do we need the _t suffix? Doesn't add any useful information AFAICT?

mlyle, 2018-03-18:
No, _t predates stdint, etc.  It meant "type"-- it puts things into a different namespace from identifiers, makes things easy to grep for, etc.  Now the OS is kinda claiming it and some people agree, some people don't.