d-ronin/dRonin #1968: Output channel form is baroque

mlyle, 2017-10-08:
Seen on current next.  "save" doesn't seem to commit changes.

Witnessed this with instrumentation on current build.  Clicking "Save"--
```
15:12:20[debug][/home/mlyle/dRonin/ground/gcs/src/plugins/uavobjectwidgetutils/smartsavebutton.cpp:243 void smartSaveButton::processOperation(QPushButton*, bool)] RESULT "All operations finished successfully" () () ()
```

But sim didn't record objectpersistence as changing.  Changing actuator settings and clicking save in uavo browser did persist properly.

Watching in UAVO browser, it looks like ActuatorSettings changes during the click, but not ObjectPersistence.  On other panes, objectpersistence does change.

mlyle, 2017-10-08:
Reverted a bunch of UI changes and it seems to work:

*    This reverts commit 0b03e1ec97b8242ff6957e9fe6f5eb42a4086f0f.
*    This reverts commit ea7ac6cbac4e3aba7cd4fdb497d3017bfb806a23.
*    This reverts commit 907bed57084af978f378567785f97f6ef77af25a.
*    This reverts commit 4e1fe3f390fd5bf84d1792e30113a31dfcff4552.
*    This reverts commit d97fee238c9a1faa9e91f3b70d7e7e2beae8456e.
*    This reverts commit 4e1fe3f390fd5bf84d1792e30113a31dfcff4552.
*    This reverts commit d97fee238c9a1faa9e91f3b70d7e7e2beae8456e.
*    This reverts commit 85da755ad4f27548c7b78979090e91d0aa5a1d5e.
*    This reverts commit d60a035014d612cba3b5aa3ccc5b04f28c50c6a5.

@tracernz I think this is your code :/

tracernz, 2017-10-08:
Just https://github.com/d-ronin/dRonin/commit/4e1fe3f390fd5bf84d1792e30113a31dfcff4552 ? (note TODOs)

tracernz, 2017-10-08:
Will look at it some time this week.

PR #1969 (2017-10-09): GCS:Config: Hax ConfigOutputWidget to save the UAVObject
TODO: Rewrite it in modern style, and then make the ConfigTaskWidget functions non-virtual so nobody can ever override them.

fixes #1968.
closed 2017-10-09 by mlyle